campreilly/UnderSeaModelingLibrary #95: seq_vector not working correctly with BOOST_FOREACH

campreilly, 2015-01-09:
-----Original Message-----
From: Kenney, Michael S CIV NAWCTSD, 4.6.8.8 [mailto:Michael.Kenney@navy.mil] 
Sent: Thursday, January 08, 2015 11:38 AM
To: Sean Reilly
Cc: Ted Burns
Subject: RE: Upgrading Ocean Acoustic Modeling

Sean,
	Here is an example:
	The code below is just used to write the data to a file and what I had to do to change it to work now. Look at the commented out BOOST_FOREACH statements

R/
MIke
```
case BTH_GENERATED:
	{
		// type cast the message in the the appropriate type
		CDogBathymetryResultMessage *pBTHResultMessage = (CDogBathymetryResultMessage*) pMessage;

		std::cout << "bth node count: " << (pBTHResultMessage->getBathymetryCollection())->size() << std::endl;
		CBathymetryCollection *pCollection = pBTHResultMessage->getBathymetryCollection();

		if (m_bUseDataGrid)
		{
			usml::types::data_grid<double, 2> *pGrid = NULL;

			pGrid = pCollection->exportTo2DDataGrid();
			size_t index[2] = { 0, 0 };
			double depth = 0.0f;

			std::ofstream bthFile;
			QString outputFile;
			outputFile = lineEditDestinationDir->text();
			if (outputFile[outputFile.length() - 1] != '/')
			{
				outputFile += "/";
			}
			outputFile += "BTH_DATA.grid";
			bthFile.open(outputFile.toAscii(), std::ios::out);

			if (!bthFile.bad() && pGrid != NULL)
			{
				bthFile.precision(4);
				bthFile.setf(std::ios::fixed);

				// BOOST_FOREACH(const double &lat, *pGrid->axis(0))
				for (size_t lat_index = 0; lat_index < pGrid->axis(0)->size(); lat_index++)
				{

					//BOOST_FOREACH(const double &lon, *pGrid->axis(1))
					for (size_t lon_index = 0; lon_index < pGrid->axis(1)->size(); lon_index++)
					{

						depth = pGrid->data(index);
						bthFile << "Lat: " << pGrid->axis(0)[lat_index] << " \tLon: " << pGrid->axis(1)[lon_index] << " \tDepth: " << depth << std::endl;
						index[1]++;
					}
					index[0]++;
					index[1] = 0;
				}
				delete pGrid;
				pGrid = NULL;
				bthFile.close();
			}
		}

		delete pBTHResultMessage;
		break;
	}
```

campreilly, 2015-01-09:
Not time critical, but may be a sign of other problems.  So we should keep track of it.

Tibonium, 2015-03-04:
Defining the following types resolves this issue with seq_linear.

<code>template<></code>
<code>struct range_mutable_iterator< usml::types::seq_linear > {</code>
<code>     typedef usml::types::seq_vector::const_iterator  type ;</code>
<code>};</code>  
  
<code>template<></code>
<code>struct range_const_iterator< usml::types::seq_linear > {</code>
<code>     typedef usml::types::seq_vector::const_iterator  type ;</code>
<code>}; </code>

Similar specialized template typedef would resolve this issue for seq_data, etc.

campreilly, 2015-03-08:
I'd like to go ahead and add these template specializations for all of the seq_vector sub-classes.

Tibonium, 2015-03-09:
Issue resolved with commit 8688fa91b3edbe4fd333d48edb9ad4c5638cfee9

campreilly, 2015-04-07:
Having problems getting the update seq_vector classes working with LVCMS.  Probably something minor.  Re-opening until we can figure out what's wrong.

campreilly, 2015-05-06:
All of the tests run at NAWCTSD now pas using release 1.0

closed 2015-05-06 by campreilly