campreilly/UnderSeaModelingLibrary #108: data_grid axis of seq_log

Tibonium, 2015-02-10:
A data_grid's axes may be any type of seq_vector, including seq_log. When a data_grid is constructed using a seq_log axis, we expect that all types of interpolation work as expected. This includes that when interpolation is called on an axis value for which there is a data point in the grid, we would expect that data point to be returned. However, we have noted that when using a seq_log axis, with either PCHIP and LINEAR interpolation, this is not the case.

Tibonium, 2015-03-16:
Does data_grid work as expected when the axes are anything other than seq_linear? We haven't every tried to create a data_grid using axes other than seq_linear. This bug could be an inherent relation to the spacing of the elements in the grid in absolute space and not indexing space.

campreilly, 2023-09-22:
Need to check of the accuracy of the derivatives calculations.  The errors in deriv_1d_test are bigger than expected, but they also seems to be big in master branch.  BOOST_CHECK comments our in master.