campreilly/UnderSeaModelingLibrary #114: Wave_queue unpredictable in Win x64 Release mode

Tibonium, 2015-03-04:
When running usml_test using a Win x64 release build the reflection_test/reflect_grid_test fails. The latitude and longitude are significantly different from the expected values. After further inspection, it was discovered that the wavefront produced in this test and the malta_movie are bogus wave propagations.  They appear to penetrate the bottom of the ocean and continue to propagate, It appears to propagate in a correct manner, sometimes, but are inaccurate portrayal of real world propagation.

It was further discovered that this only occurs when using a Win64 release build. The Win64 debug, Win32 release/debug, or any linux build, does not produce these errors.  Further investigation is needed to determine the full extent of this unpredictable propagation nature.

campreilly, 2015-03-15:
I finally got my Win64 environment up and running, and I consistently see this same problem.  Here is what I've learned so far.

When I look at the CSV files that are output by reflection_test/reflect_grid_test, it looks like the ray is failing to detect a collision with the bottom in Release mode, but it works fine in Debug mode.

Debug Mode
[image]

Release Mode:
[image]

Up until 7.1 seconds, the CSV files are identical.  A refection should occur between 7.1 and 7.2 sec, but in release mode, the direction of the ray is not changed, and it keeps heading deeper into the bottom. 

At some points, I thought I saw DE and AZ values that were different in Release than in Debug.  But I've been unable to reproduce those results.  Did I just imagine them?

Tibonium, 2015-03-15:
No, you are not mistaken. The incorrect launch angles were produced on my machine. With DE at -89~ and AZ at -180.

campreilly, 2015-03-15:
More info:  I added a debugging print statement into wave_queue::detect_reflections_bottom()
just before ```if ( depth > 0.0 )``` on line 235.  It does not seem that ```_ocean.bottom().height()``` is passing back a valid depth.  Note below how the heights in Debug mode are all around 6.36e+6.

Debug mode:
```
time=0 height=6.36775e+006 rho=6.37143e+006 depth=-3674.78
time=0.1 height=6.36776e+006 rho=6.37138e+006 depth=-3620.57
time=0.2 height=6.36776e+006 rho=6.37132e+006 depth=-3566.36
time=0.3 height=6.36776e+006 rho=6.37127e+006 depth=-3512.16
time=0.4 height=6.36776e+006 rho=6.37122e+006 depth=-3457.96
time=0.5 height=6.36777e+006 rho=6.37117e+006 depth=-3403.76
time=0.6 height=6.36777e+006 rho=6.37112e+006 depth=-3349.56
time=0.7 height=6.36777e+006 rho=6.37107e+006 depth=-3295.37
time=0.8 height=6.36778e+006 rho=6.37102e+006 depth=-3241.18
time=0.9 height=6.36778e+006 rho=6.37097e+006 depth=-3186.99
time=1 height=6.36778e+006 rho=6.37091e+006 depth=-3134.71
time=1.1 height=6.36778e+006 rho=6.37086e+006 depth=-3085.79
time=1.2 height=6.36778e+006 rho=6.37081e+006 depth=-3036.87
time=1.3 height=6.36777e+006 rho=6.37076e+006 depth=-2987.96
time=1.4 height=6.36777e+006 rho=6.37071e+006 depth=-2939.06
time=1.5 height=6.36777e+006 rho=6.37066e+006 depth=-2890.15
time=1.6 height=6.36777e+006 rho=6.37061e+006 depth=-2841.25
time=1.7 height=6.36776e+006 rho=6.37056e+006 depth=-2792.35
time=1.8 height=6.36776e+006 rho=6.3705e+006 depth=-2743.46
time=1.9 height=6.36776e+006 rho=6.37045e+006 depth=-2694.56
time=2 height=6.36776e+006 rho=6.3704e+006 depth=-2645.67
time=2.1 height=6.36776e+006 rho=6.37035e+006 depth=-2593.04
time=2.2 height=6.36776e+006 rho=6.3703e+006 depth=-2538.99
time=2.3 height=6.36776e+006 rho=6.37025e+006 depth=-2484.94
time=2.4 height=6.36777e+006 rho=6.3702e+006 depth=-2430.9
time=2.5 height=6.36777e+006 rho=6.37015e+006 depth=-2376.85
time=2.6 height=6.36777e+006 rho=6.37009e+006 depth=-2322.81
time=2.7 height=6.36777e+006 rho=6.37004e+006 depth=-2268.77
time=2.8 height=6.36778e+006 rho=6.36999e+006 depth=-2214.74
time=2.9 height=6.36778e+006 rho=6.36994e+006 depth=-2160.71
time=3 height=6.36778e+006 rho=6.36989e+006 depth=-2106.68
time=3.1 height=6.36779e+006 rho=6.36984e+006 depth=-2052.92
time=3.2 height=6.36779e+006 rho=6.36979e+006 depth=-2001.81
time=3.3 height=6.36779e+006 rho=6.36974e+006 depth=-1950.7
time=3.4 height=6.36779e+006 rho=6.36968e+006 depth=-1899.6
time=3.5 height=6.36779e+006 rho=6.36963e+006 depth=-1848.5
time=3.6 height=6.36779e+006 rho=6.36958e+006 depth=-1797.4
time=3.7 height=6.36779e+006 rho=6.36953e+006 depth=-1746.31
time=3.8 height=6.36778e+006 rho=6.36948e+006 depth=-1695.22
time=3.9 height=6.36778e+006 rho=6.36943e+006 depth=-1644.13
time=4 height=6.36778e+006 rho=6.36938e+006 depth=-1593.04
time=4.1 height=6.36778e+006 rho=6.36933e+006 depth=-1541.96
time=4.2 height=6.36779e+006 rho=6.36928e+006 depth=-1489.47
time=4.3 height=6.36779e+006 rho=6.36922e+006 depth=-1435.3
time=4.4 height=6.36779e+006 rho=6.36917e+006 depth=-1381.12
time=4.5 height=6.3678e+006 rho=6.36912e+006 depth=-1326.95
time=4.6 height=6.3678e+006 rho=6.36907e+006 depth=-1272.79
time=4.7 height=6.3678e+006 rho=6.36902e+006 depth=-1218.63
time=4.8 height=6.3678e+006 rho=6.36897e+006 depth=-1164.47
time=4.9 height=6.36781e+006 rho=6.36892e+006 depth=-1110.31
time=5 height=6.36781e+006 rho=6.36887e+006 depth=-1056.16
time=5.1 height=6.36781e+006 rho=6.36882e+006 depth=-1002.01
time=5.2 height=6.36782e+006 rho=6.36876e+006 depth=-947.86
time=5.3 height=6.36782e+006 rho=6.36871e+006 depth=-896.345
time=5.4 height=6.36782e+006 rho=6.36866e+006 depth=-845.399
time=5.5 height=6.36782e+006 rho=6.36861e+006 depth=-794.455
time=5.6 height=6.36782e+006 rho=6.36856e+006 depth=-743.514
time=5.7 height=6.36782e+006 rho=6.36851e+006 depth=-692.576
time=5.8 height=6.36782e+006 rho=6.36846e+006 depth=-641.641
time=5.9 height=6.36782e+006 rho=6.36841e+006 depth=-590.708
time=6 height=6.36782e+006 rho=6.36835e+006 depth=-539.779
time=6.1 height=6.36781e+006 rho=6.3683e+006 depth=-488.853
time=6.2 height=6.36781e+006 rho=6.36825e+006 depth=-437.93
time=6.3 height=6.36782e+006 rho=6.3682e+006 depth=-386.592
time=6.4 height=6.36782e+006 rho=6.36815e+006 depth=-333.515
[... 68 lines cut ...]
time=5.2 height=105959 rho=6.36876e+006 depth=-6.2628e+006
time=5.3 height=5.87478e+006 rho=6.36871e+006 depth=-493934
time=5.4 height=5.27577e+006 rho=6.36866e+006 depth=-1.09289e+006
time=5.5 height=4.67675e+006 rho=6.36861e+006 depth=-1.69186e+006
time=5.6 height=4.07773e+006 rho=6.36856e+006 depth=-2.29083e+006
time=5.7 height=3.47869e+006 rho=6.36851e+006 depth=-2.88982e+006
time=5.8 height=2.87965e+006 rho=6.36846e+006 depth=-3.48881e+006
time=5.9 height=2.28059e+006 rho=6.36841e+006 depth=-4.08782e+006
time=6 height=1.68153e+006 rho=6.36835e+006 depth=-4.68683e+006
time=6.1 height=1.08245e+006 rho=6.3683e+006 depth=-5.28585e+006
time=6.2 height=483367 rho=6.36825e+006 depth=-5.88489e+006
time=6.3 height=6.25209e+006 rho=6.3682e+006 depth=-116114
time=6.4 height=5.65298e+006 rho=6.36815e+006 depth=-715166
time=6.5 height=5.05387e+006 rho=6.3681e+006 depth=-1.31423e+006
time=6.6 height=4.45475e+006 rho=6.36805e+006 depth=-1.9133e+006
time=6.7 height=3.85562e+006 rho=6.368e+006 depth=-2.51238e+006
time=6.8 height=3.25648e+006 rho=6.36795e+006 depth=-3.11147e+006
time=6.9 height=2.65733e+006 rho=6.3679e+006 depth=-3.71057e+006
time=7 height=2.05816e+006 rho=6.36784e+006 depth=-4.30968e+006
time=7.1 height=1.45899e+006 rho=6.36779e+006 depth=-4.9088e+006
time=7.2 height=859814 rho=6.36774e+006 depth=-5.50793e+006
time=7.3 height=260625 rho=6.36769e+006 depth=-6.10707e+006
time=7.4 height=6.02926e+006 rho=6.36764e+006 depth=-338380
time=7.5 height=5.43005e+006 rho=6.36759e+006 depth=-937539
```

Tibonium, 2015-03-16:
Figured out what is causing the error, but have not determined why it is happening. In the call for detect_bottom_reflection we were doing a "fast" computation of the bottom height by using a linear interpolation, so we passed a true to the height call. Changing the call to false and hence using a PCHIP, resolved the error.

Tibonium, 2015-03-16:
In attempt to flush out the error, I added cout debug statements to the data_grid::linear call. However, when the test is run the expected results are produced and hence no errors...

Tibonium, 2015-03-16:
Just as a cautionary check, I checked out a branch prior to the redesign of seq_vector and ran my specialized test. The same results are produced. Linear interp produces bogus results while pchip/nearest produce correct results.

campreilly, 2015-03-17:
We've narrowed this down to the following section of code.

```
        DATA_TYPE linear(int dim, const size_t* index, const double* location,
                DATA_TYPE& deriv, DATA_TYPE* deriv_vec) const
        {
            DATA_TYPE result, da, db;

            // build interpolation coefficients

	    size_t next[NUM_DIMS];
            memcpy(next, index, NUM_DIMS * sizeof(size_t));
            ++next[dim];

            const DATA_TYPE a = interp(dim - 1, index, location, da, deriv_vec);
            const DATA_TYPE b = interp(dim - 1, next, location, db, deriv_vec);
            const size_t k = index[dim];
            seq_vector* ax = _axis[dim];
            ....
```
If we add debug cout statements after the ```const DATA_TYPE a = interp()``` and ```const DATA_TYPE b = interp()``` we can see that *a* changes value while *b* is being computed.  Even though linear is used recursively, *a* is a local variable and this should not happen.

It only seems to happen for linear 2-D interpolation of ETOPO1 bathymetry, and only using the Release x64 model of Visual C+++ 2012 in Visual Studio 2013.  Perhaps it has something to do with optimizing a recursive template function.

As a workaround, it seems to stop happening if we replace 
```
size_t next[NUM_DIMS];
```
with 

```
size_t* next = new size_t[NUM_DIMS];
blah, blah, blah...
delete[] next;
```

But, that is just a work around.

Tibonium, 2015-03-18:
I had an epiphany last night. If the problem resembles a dynamic memory allocation issue then perhaps a reordering of the events would resolve the issue. So I rearranged the code as below:

<code>const DATA_TYPE a = interp(dim - 1, index, location, da, deriv_vec);</code>
<code>size_t next[NUM_DIMS];</code>
<code>memcpy(next, index, NUM_DIMS * sizeof(size_t));</code>
<code>++next[dim];</code>
<code>const DATA_TYPE b = interp(dim - 1, next, location, db, deriv_vec);</code>

Rerunning the test, grid_2d_test showed that the bogus values were no longer being produced. I can only conjecture at this point as to the fundamental cause at this bizarre situation and so I can only provide solutions and not answers.

commit 435adfae (2015-03-18): Resolves #114, it appears this was missed in the initial cut for release1.0
closed 2015-03-18 by Tibonium