campreilly/UnderSeaModelingLibrary #115: Seq_vector virtual operator()

Tibonium, 2015-03-15:
After the redesign of seq_vector a big occurred when tests used the data_grid pchip calculation, the results that were returned were bogus. After further investigation it was found that the redesigned seq_vector class had dropped the keyword virtual from the operator (). Adding back the virtual keyword resolved the bogus values.

Originally, the operator () was a virtual function so the children class could provide specialized methods but found that live computing for some too to much time, and so the operator () was supplied from all children classes but the virtual keyword remained. It was find that without the virtual keyword the same errors showed up in data_grid pchip.

Why does the operator () needed a virtual keyword in seq_vector?

Tibonium, 2015-05-22:
This appears to be an optimization problem. Reran tests without virtual keyword in debug mode and no errors were detected. Then tested in release build and the previous error began to occur in the regression tests.

Tibonium, 2015-05-28:
In an attempt to reproduce this in multiple ways, I created a standalone program and linked usml release build to it. Copying the exact test of cubic_1d_test for initial testing, I found that my standalone program produces valid and accurate results.....but the usml_test version produces erroneous results. The usml_test and standalone version are identical in implementation. as provided below:

````
#define BOOST_TEST_DYN_LINK
#define BOOST_TEST_MODULE main
#include <boost/test/unit_test.hpp>

#include <usml/types/types.h>
#include <iostream>

using namespace std ;
using namespace usml::types ;

double cubic_1d( double x ) {
    return 5.0 + 3.0*x - 0.3*x*x ;
}

BOOST_AUTO_TEST_CASE( virtual_keyword )
{

    double n = 9.0 ;
    seq_linear x_axis(1.0, 2.0, n) ;
    const seq_vector* axis[] = { &x_axis } ;
    data_grid<double,1> grid( axis ) ;
    
    for(size_t i=0; i<x_axis.size(); ++i)
        grid.data( &i, cubic_1d(x_axis(i)) ) ;
        
    grid.interp_type(0, GRID_INTERP_PCHIP) ;
    
    char buff[256] ;
    sprintf(buff,"x%*ctruth%*cpchip",8,' ',6,' ') ;
    cout << buff << endl ;
    for(double x=1.0; x<=n; x+=0.25) {
        double truth = cubic_1d(x) ;
        double pchip = grid.interpolate(&x) ;
        sprintf(buff,"%1.3f  %8.3f  %8.3f",x,truth,pchip) ;
        cout << buff << endl ;
        BOOST_CHECK_CLOSE( truth, pchip, 2.0 ) ;
    }

}   // end of BOOST_AUTO_TEST_CASE
````

Tibonium, 2015-05-28:
On a side note, the we state that the test is utilizing a cubic equation to populate the data inside of the grid, but the equation is only a second degree equation and hence a quadratic, not cubic (third degree).

campreilly, 2023-09-22:
Checked to see that C++17 build gets same answers as test shown in this issue.  But update data_grid_test to change "cubic" function to "quadratic".

closed 2023-09-22 by campreilly