campreilly/UnderSeaModelingLibrary #128: Frequency response of beam_pattern_sine and beam_pattern_cosine

campreilly, 2015-04-01:
Currently these beam patterns do not appear to include a frequency dependence.  If we can assume that these patterns are supposed to model the response of a dipole array.  Then the response should be

Cosine: B(a) = cos[ 2*pi*f / c * cos(a) ]
Sine: B(a) = cos[ 2*pi*f / c * sin(a) ]

a = conic angle from forward endfire
f = frequency
c = speed of sound

Reference: http://www.personal.psu.edu/faculty/m/x/mxm14/sonar/beamforming.pdf

Tibonium, 2015-04-03:
Is there any reason that a beam_pattern_line with 2 elements is not sufficient enough?

Tibonium, 2015-04-10:
If it would be best, similar to HLA/VLA classes, I can create two classes that are specialized for both of these functions. Also, if I understand correctly, there seems to be a missing factor of d (spacing) in both of the equations your provided.

campreilly, 2015-04-10:
Yes, I looked at the reference again, and the "d" is clearly missing.

I re-looked at the Production Sonobuoy spec, and Figure 6 clearly says that the Cosine channel has its MRA along the reference axis (north-south) and that the Sine is perpendicular.

campreilly, 2015-05-06:
It turns out that the sine and cosine beams are not frequencies depdendent.  But they also do not support a null that goes down to zero.  Changes the title of this issue to match the work remaining.

Tibonium, 2015-05-12:
What is the remaining work to be done?
I see two things started as follows:
* Implement frequency dependent Sine/Cosine beams
 * The spec file shows that they are frequency independent, so is this still required?
* Implement Cosine/Sine as the directional pattern designated in the spec file
 * This is already done for the frequency independent beam patterns provided
* No Null?
 * I don't know what you're asking for this, do you have a beam pattern grid that supports this?

campreilly, 2015-05-18:
Additional work is required.

campreilly, 2015-05-19:
A copy of the design requirements can be found in C:\Development\Projects\usml_private\software_process\designs\Sine-Cosine Beam Directions.docx .
Here is the text from that document:

1) Create a beam_pattern_model for the cosine/sine beam that obeys the equation

[image]

Include the same ability to modify the orientation of this array that we use in our other beam_pattern_model classes.

Demonstrate that for the special case of L = -10 dB, b(θ,ϕ,f,L,ϕ_r )=1/2 when ϕ-ϕ_r=63.6 degrees.

2) Test this beam pattern solution by plotting a 3-D surface in Matlab.

[image]

Demonstrate that the beam pattern dips down in the northeast quadrant when a tilt of 20 degrees is applied in the 45 degree direction.

3) Compute the analytic solution for the directivity index of the cosine/sine beam.

[image]

Implement this directivity index as part of the beam_pattern_model for the cosine/sine beams.  You can assume that the directivity index is impendent of array orientation.  Demonstrate that in the special case where L = -300 dB, the beam_pattern_model produces a directivity index such that DI(f,L)=10 log10(3) 

4) Update the doxygen comments in the code and include a discussion of equations 1, 3, and 4 in those comments.

5) Update the unit tests to reflect these changes.

PR #209 (2015-06-11): Updated Sine and Cosine beam patterns
Resolves #128 
Added the ability to customize the cosine/sine pattern's null value and gain factor.
closed 2015-06-11 by campreilly