campreilly/UnderSeaModelingLibrary #130: Compute eigenverb widths as a function of freqency

campreilly, 2015-04-01:
Eigenverbs are the projection of Gaussian beams onto an interface during a collision.  The Gaussian beams have a frequency dependent height/width, given by eq. (85) in the theory paper

(w')^2 = (2w)^2 + (2 pi wavelength)^2

The eigenverbs also need to a frequency dependent height/width.

campreilly, 2015-04-09:
Actively working this issue on the eigenverb_widths branch.

campreilly, 2015-04-10:
The eigenverb generation on the eigenverb_widths branch is working pretty darn well but:
- Should I wait to check in until I've proven the results of eigenverb_basic are accurate?
- Should I wait to check in until I've updated the envelope calculations to use these changes?  Currently most of the envelope modeling is stubbed out.
- How soon is the rest of the team going to need these changes so that they can work on the "compute propagation effects" use case (aka adding eigenverbs to the fathometer work)?
- Why do I keep forgetting to reference this issue in my commit messages?  Am I just a knucklehead?

Tibonium, 2015-04-20:
I think it would be best to have the accuracy and the newly updated code prior to committing to master. As they are rather large changes to the overall structure of the envelopes, I feel that having this updated and ready to go is a good idea.

campreilly, 2015-04-20:
The only downside to waiting is that we are unable to make progress on the eigenverb part of wavefront_generator until I work this out.  But, that is something I can work on today.

campreilly, 2015-04-20:
Resolved by pull request #158 .

campreilly, 2015-06-04:
I believe that we need to revert these changes until more work can be done on the theory.

We built a new version of eigenverb_demo.m, called eigenverb_freq.m, to prototype the effects of overlap and frequency spreading, but can't get the normalization to come out correctly.
    - Figured out how to built Es and Er in terms of init_area and absorption loss.
    - Like eigenverb_model.m there is an extra factor of 4 pi in Esr denominator.
    - There is an extra factor of "overlap" in Esr denominator.
    - Direct path is still high, this appears to be the result of problems in normalizing the frequency part of spreading.

Because the new eigenverbs are much wider than those without a 2x overlap, this algorithm is also much slower than the one with frequency dependent spreading.  And it is not clear that all that extra work is really paying off in fidelity.  Here is the results of eigenverb_freq.m with overlap=1 and frequency=1e6 Hz (no spreading).
[image]
Here are the results with  overlap=2 and frequency=100 Hz (about 100 meters of spreading)..
[image]
Near each initial impact, the spread version is clearly fatter than the unspread version.  In the area between 1 and 4 seconds, details in the total reverberation have been washed out by the broader beams.  Both of these are expected, but can we really say this is a higher resolution result?  Is all the extra work worth it?

Although I think we need to deal with this everntually, I think we should move it out of the current reverberation delivery and mark it as an "idea for later".

closed 2023-09-20 by campreilly