campreilly/UnderSeaModelingLibrary #131: beam_pattern_grid won't compile in Linux with pedantic option turned on.

campreilly, 2015-04-01:
Apparently the compiler doen't lik the way that data copying is being done n construct_pattern
Lines 203-241
```
        /**
         * Constructs the data grid using the passed in data.
         */
        void construct_pattern( const value_ptr data,
                                const data_units& data_unit )
        {
            size_type N = 1 ;
            for(size_type i=0; i<Dim; ++i) {
                N *= this->_axis[i]->size() ;
                this->interp_type( i, GRID_INTERP_PCHIP ) ;
            }
            switch( data_unit ) {

                /**
                 * Data that has been passed in is in log units
                 * and needs to be converted to linear units
                 */
                case LOG_UNITS :
                {
                    value_ptr data_copy = new value_type[N] ;
                    memcpy( data_copy, data, N*sizeof(value_type) ) ;
                    for(size_type i=0; i<N; ++i) {
                        *(data_copy++) = std::pow( 10.0, (-(*data_copy)/10.0) ) ;
                    }
                    memcpy( this->_data, data_copy, N*sizeof(value_type) ) ;
                    delete data_copy ;
                    break ;
                }

                /**
                 * The data was already passed to this constructor in
                 * linear units.
                 */
                default:
                    memcpy( this->_data, data, N*sizeof(value_type) ) ;
                    break ;
            }
            memset( this->_edge_limit, true, Dim*sizeof(bool) ) ;
        }

```

This results in:

```
/usr/bin/c++   -DBOOST_ALL_DYN_LINK -DBOOST_UBLAS_CHECK_DIVISION_TYPE -DUSML_DATA_DIR=\"/home/sreilly/Projects/usml/data\" -DUSML_TEST_DIR=\"/home/sreilly/Projects/usml\" -O3 -DNDEBUG -I/home/sreilly/Projects/usml/..    -g -ffast-math -fno-finite-math-only -std=c++98 -pedantic -Wall -Werror -Wno-long-long -Wno-sign-compare -o CMakeFiles/usml_test.dir/sensors/test/beam_pattern_test.cc.o -c /home/sreilly/Projects/usml/sensors/test/beam_pattern_test.cc
cc1plus: warnings being treated as errors
In file included from /home/sreilly/Projects/usml/../usml/sensors/beams.h:23,
                 from /home/sreilly/Projects/usml/sensors/test/beam_pattern_test.cc:6:
/home/sreilly/Projects/usml/../usml/sensors/beam_pattern_grid.h: In member function ‘void usml::sensors::beam_pattern_grid<T, Dim>::construct_pattern(T*, const usml::sensors::beam_pattern_grid<T, Dim>::data_units&) [with T = double, long unsigned int Dim = 1ul]’:
/home/sreilly/Projects/usml/../usml/sensors/beam_pattern_grid.h:56:   instantiated from ‘usml::sensors::beam_pattern_grid<T, Dim>::beam_pattern_grid(usml::types::seq_vector**, T*, const usml::sensors::beam_pattern_grid<T, Dim>::data_units&) [with T = double, long unsigned int Dim = 1ul]’
/home/sreilly/Projects/usml/sensors/test/beam_pattern_test.cc:384:   instantiated from here
/home/sreilly/Projects/usml/../usml/sensors/beam_pattern_grid.h:225: error: operation on ‘data_copy’ may be undefined
/home/sreilly/Projects/usml/../usml/sensors/beam_pattern_grid.h: In member function ‘void usml::sensors::beam_pattern_grid<T, Dim>::construct_pattern(T*, const usml::sensors::beam_pattern_grid<T, Dim>::data_units&) [with T = double, long unsigned int Dim = 2ul]’:
/home/sreilly/Projects/usml/../usml/sensors/beam_pattern_grid.h:56:   instantiated from ‘usml::sensors::beam_pattern_grid<T, Dim>::beam_pattern_grid(usml::types::seq_vector**, T*, const usml::sensors::beam_pattern_grid<T, Dim>::data_units&) [with T = double, long unsigned int Dim = 2ul]’
/home/sreilly/Projects/usml/sensors/test/beam_pattern_test.cc:441:   instantiated from here
/home/sreilly/Projects/usml/../usml/sensors/beam_pattern_grid.h:225: error: operation on ‘data_copy’ may be undefined
/home/sreilly/Projects/usml/../usml/sensors/beam_pattern_grid.h: In member function ‘void usml::sensors::beam_pattern_grid<T, Dim>::construct_pattern(T*, const usml::sensors::beam_pattern_grid<T, Dim>::data_units&) [with T = double, long unsigned int Dim = 3ul]’:
/home/sreilly/Projects/usml/../usml/sensors/beam_pattern_grid.h:56:   instantiated from ‘usml::sensors::beam_pattern_grid<T, Dim>::beam_pattern_grid(usml::types::seq_vector**, T*, const usml::sensors::beam_pattern_grid<T, Dim>::data_units&) [with T = double, long unsigned int Dim = 3ul]’
/home/sreilly/Projects/usml/sensors/test/beam_pattern_test.cc:509:   instantiated from here
/home/sreilly/Projects/usml/../usml/sensors/beam_pattern_grid.h:225: error: operation on ‘data_copy’ may be undefined
gmake[2]: *** [CMakeFiles/usml_test.dir/sensors/test/beam_pattern_test.cc.o] Error 1
```

Tibonium, 2015-04-02:
I am unable to address this error as I cannot get passed wavefront_generator obj creation

[  6%] Building CXX object CMakeFiles/usml.dir/eigenverb/wavefront_generator.cc.o
cc1plus: warnings being treated as errors
/home/Tibonium/projects/usml/../usml/eigenverb/wavefront_generator.h: In constructor ‘usml::eigenverb::wavefront_generator::wavefront_generator()’:
/home/Tibonium/projects/usml/../usml/eigenverb/wavefront_generator.h:175: error: ‘usml::eigenverb::wavefront_generator::_runID’ will be initialized after
/home/Tibonium/projects/usml/../usml/eigenverb/wavefront_generator.h:167: error:   ‘bool usml::eigenverb::wavefront_generator::_done’
/home/Tibonium/projects/usml/eigenverb/wavefront_generator.cc:22: error:   when initialized here
/home/Tibonium/projects/usml/../boost/asio/error.hpp: At global scope:
/home/Tibonium/projects/usml/../boost/asio/error.hpp:252: error: ‘boost::asio::error::system_category’ defined but not used
/home/Tibonium/projects/usml/../boost/asio/error.hpp:254: error: ‘boost::asio::error::netdb_category’ defined but not used
/home/Tibonium/projects/usml/../boost/asio/error.hpp:256: error: ‘boost::asio::error::addrinfo_category’ defined but not used
/home/Tibonium/projects/usml/../boost/asio/error.hpp:258: error: ‘boost::asio::error::misc_category’ defined but not used
/home/Tibonium/projects/usml/../boost/system/error_code.hpp:222: error: ‘boost::system::posix_category’ defined but not used
/home/Tibonium/projects/usml/../boost/system/error_code.hpp:223: error: ‘boost::system::errno_ecat’ defined but not used
/home/Tibonium/projects/usml/../boost/system/error_code.hpp:224: error: ‘boost::system::native_ecat’ defined but not used
make[2]: *** [CMakeFiles/usml.dir/eigenverb/wavefront_generator.cc.o] Error 1
make[1]: *** [CMakeFiles/usml.dir/all] Error 2

Tibonium, 2015-04-03:
Linux was complaining about the operator++:

````
//                        *(data_copy++) = std::pow( 10.0, (-(*data_copy)/10.0) ) ;
                        data_copy[i] = std::pow( 10.0, (-data_copy[i]/10.0) ) ;
````
warning goes away with the above change, but the original format is valid.

commit 3e00ff71 (2015-04-06): Resolves #131
closed 2015-04-06 by Tibonium