campreilly/UnderSeaModelingLibrary #132: Fix sensor_model::run_wave_generator problems created by smart_ptr work

campreilly, 2015-04-02:
- The ```sensor_model::run_wave_generator()``` was stubbed out during smart pointer work because it was crashing. Re-enable and fix it.  
- To support stand alone testing, ensure that is does not attempt to build a ```wavefront_generator``` if ```ocean_shared::current().get() == NULL```.

campreilly, 2015-04-06:
I haven't seen any updates that show that Ted has fixed this issue yet.

Tibonium, 2015-04-06:
I think you need to double check your pulls because this resolution is in there....
````
void sensor_model::run_wave_generator() {

    // Only run wavefront generator if ocean_model pointer is not NULL
    if (ocean_shared::current().get() != NULL ) {

        #ifdef USML_DEBUG
            cout << "sensor: run_wave_generator(" << sensorID() << ")" << endl ;
        #endif
        // Create the wavefront_generator
        wavefront_generator* generator = new wavefront_generator();

        // Populate waveq3d settings
        generator->wavefront_listener(this);
        generator->sensor_position(_position);

        // Get the targets sensor references
        std::list<sensor_model::reference> targets = sensor_targets();

        // Get the target positions
        wposition target_pos = target_positions(targets);
        generator->targets(target_pos);

        // Set the targetID's for later use in sending on to sensor_pair
        target_ids(targets);

        // Get the frequencies
        generator->frequencies(_source->frequencies());

        // Set the ocean_model
        generator->ocean(ocean_shared::current());

        // Make wavefront_generator a wavefront_task, with use of shared_ptr
        _wavefront_task = thread_task::reference(generator);

        // Pass in to thread_pool
        thread_controller::instance()->run(_wavefront_task);
    }

    #ifdef USML_DEBUG
        if (ocean_shared::current().get() == NULL ) {
             cout << "sensor: run_wave_generator no ocean provided !!! (" << sensorID() << ")" << endl ;
        }
    #endif
}
````

commit 38805284 (2015-04-03): Merge pull request #126 from campreilly/wavefront_generator

Wavefront generator resolves #132, #133, #134
closed 2015-04-06 by Tibonium