campreilly/UnderSeaModelingLibrary #140: Add thread support to query or poll thread_pool still has thread running

islanderjtn, 2015-04-07:
thread_pool needs a call added to see if thread id is still running.

campreilly, 2015-04-07:
Or, alternatively if any thread_tasks are still waiting to be completed.  This would allow our unit tests to wait until the thread_controller was done with processing before exiting the main thread.  Exiting the main thread currently aborts any thread_tasks that are still waiting to be completed.

campreilly, 2015-04-07:
A Google search on "asio io_service wait until all tasks complete" suggests that there may be a way to do this using poll() instead of run().  But, I don't see how to make io_service::work interoperate with poll().  This may take several days of search, trial, and error.

islanderjtn, 2015-04-16:
Implemented  a num_active call to determine if any threads are still running.  See #150.

closed 2015-04-16 by islanderjtn