campreilly/UnderSeaModelingLibrary #141: Fathometer accuracy problems in reverb_analytic_test

campreilly, 2015-04-09:
The WaveQ3D model does not seem to be accurately computing eigenrays in 200 meters of water when the source is the same as the target.

Scenario:
- bottom flat 200 meter, sand
- sound speed isovelocity 1500 m/s
- time_step = 0.001
- time_maximum = 1.0
- number_de = 181
- number_az = 18
- source depth 10 meters
- only target is the source

Turn on DEBUG_OUTPUT_EIGENRAYS in wave_queue.cc to see eigenrays being created
```
wave_queue::build_eigenray() 
	target(0,0):
	t=0.0126707 inten=[1](28.4249) de=80.939 az=160
	surface=1 bottom=0 caustic=0
wave_queue::build_eigenray() 
	target(0,0):
	t=0.253438 inten=[1](63.3714) de=80.939 az=160
	surface=1 bottom=1 caustic=0
wave_queue::build_eigenray() 
	target(0,0):
	t=0.494203 inten=[1](79.226) de=-77.9156 az=160
	surface=1 bottom=2 caustic=0
wave_queue::build_eigenray() 
	target(0,0):
	t=0.506875 inten=[1](78.2781) de=80.939 az=160
	surface=2 bottom=2 caustic=0
```
Problems:
- Eigenrays are missing for surface=0 bottom=1 path
- There should be two eigenrays for surface=1 bottom=1 path, missing the one in which de is negative.
- DE angles for these paths should be much closer to +/-90 than they are.

campreilly, 2015-04-09:
Could this be related to the work that we did on Issue #36?  That issue says,

"If the target is directly above or directly below the source, it is unclear if the eigenray calculation can accurately predict the propagation loss. In this case, all of the azimuths will contribute to the Gaussian beam."

Tibonium, 2015-04-10:
Created a stand-alone test and got these results
````
=== shadow_eigenrays_test ===
creating ocean
computing max_time
propagating wavefront for 1.215
___eigenray for target(0,0) found___
	count = 1 time = 0.00316782
 	launch angles (de,az): (80.939, 0)
 	arrival angles (de,az): (80.1831, 0)
 	srf = 0 bot = 0
___eigenray for target(1,0) found___
	count = 1 time = 0.00316782
 	launch angles (de,az): (-77.9156, 0)
 	arrival angles (de,az): (-80.1831, 0)
 	srf = 0 bot = 0
___eigenray for target(0,0) found___
	count = 2 time = 0.00950694
 	launch angles (de,az): (80.939, 0)
 	arrival angles (de,az): (-80.1831, 0)
 	srf = 1 bot = 0
___eigenray for target(1,0) found___
	count = 2 time = 0.015838
 	launch angles (de,az): (80.939, 0)
 	arrival angles (de,az): (-80.1831, 0)
 	srf = 1 bot = 0
___eigenray for target(1,0) found___
	count = 3 time = 0.237597
 	launch angles (de,az): (-77.9156, 0)
 	arrival angles (de,az): (80.1831, 0)
 	srf = 0 bot = 1
___eigenray for target(0,0) found___
	count = 3 time = 0.243934
 	launch angles (de,az): (-77.9156, 0)
 	arrival angles (de,az): (80.1831, 0)
 	srf = 0 bot = 1
___eigenray for target(1,0) found___
	count = 4 time = 0.250271
 	launch angles (de,az): (80.939, 0)
 	arrival angles (de,az): (80.1831, 0)
 	srf = 1 bot = 1
___eigenray for target(0,0) found___
	count = 4 time = 0.256604
 	launch angles (de,az): (80.939, 0)
 	arrival angles (de,az): (80.1831, 0)
 	srf = 1 bot = 1
___eigenray for target(1,0) found___
	count = 5 time = 0.491033
 	launch angles (de,az): (-77.9156, 0)
 	arrival angles (de,az): (80.183, 0)
 	srf = 1 bot = 2
___eigenray for target(0,0) found___
	count = 5 time = 0.497369
 	launch angles (de,az): (-77.9156, 0)
 	arrival angles (de,az): (80.183, 0)
 	srf = 1 bot = 2
___eigenray for target(1,0) found___
	count = 6 time = 0.503703
 	launch angles (de,az): (80.939, 0)
 	arrival angles (de,az): (80.183, 0)
 	srf = 2 bot = 2
___eigenray for target(0,0) found___
	count = 6 time = 0.51004
 	launch angles (de,az): (80.939, 0)
 	arrival angles (de,az): (80.183, 0)
 	srf = 2 bot = 2
Finished propagation
````

Tibonium, 2015-04-10:
Ran again with all the same parameters that you used and placed both targets on top of the source, which appear to give the correct results. (Note DE +/-90 are not value launch angles for eigenrays)
````
=== shadow_eigenrays_test ===
creating ocean
computing max_time
propagating wavefront for 1.22
___eigenray for target(0,0) found___
	count = 1 time = 0.0126707
 	launch angles (de,az): (80.939, 0)
 	arrival angles (de,az): (-80.1831, 0)
 	srf = 1 bot = 0
___eigenray for target(1,0) found___
	count = 1 time = 0.0126707
 	launch angles (de,az): (80.939, 0)
 	arrival angles (de,az): (-80.1831, 0)
 	srf = 1 bot = 0
___eigenray for target(0,0) found___
	count = 2 time = 0.240768
 	launch angles (de,az): (-77.9156, 0)
 	arrival angles (de,az): (80.1831, 0)
 	srf = 0 bot = 1
___eigenray for target(1,0) found___
	count = 2 time = 0.240768
 	launch angles (de,az): (-77.9156, 0)
 	arrival angles (de,az): (80.1831, 0)
 	srf = 0 bot = 1
___eigenray for target(0,0) found___
	count = 3 time = 0.253438
 	launch angles (de,az): (80.939, 0)
 	arrival angles (de,az): (80.1831, 0)
 	srf = 1 bot = 1
___eigenray for target(1,0) found___
	count = 3 time = 0.253438
 	launch angles (de,az): (80.939, 0)
 	arrival angles (de,az): (80.1831, 0)
 	srf = 1 bot = 1
___eigenray for target(0,0) found___
	count = 4 time = 0.494203
 	launch angles (de,az): (-77.9156, 0)
 	arrival angles (de,az): (80.183, 0)
 	srf = 1 bot = 2
___eigenray for target(1,0) found___
	count = 4 time = 0.494203
 	launch angles (de,az): (-77.9156, 0)
 	arrival angles (de,az): (80.183, 0)
 	srf = 1 bot = 2
___eigenray for target(0,0) found___
	count = 5 time = 0.506875
 	launch angles (de,az): (80.939, 0)
 	arrival angles (de,az): (80.183, 0)
 	srf = 2 bot = 2
___eigenray for target(1,0) found___
	count = 5 time = 0.506875
 	launch angles (de,az): (80.939, 0)
 	arrival angles (de,az): (80.183, 0)
 	srf = 2 bot = 2
Finished propagation
````

Tibonium, 2015-04-13:
This appears to be an error with the setup of the wavefront. In the monostatic scenario, the De angles are heavily weighted in the horizontal direction, thus leaving very few rays in the vertical directions, +/-90 with the neighbor ray roughly +/-71. As such after only a few time steps the spacing between these rays have grown so much that eigenrays are not being found, do to the lack of neighbor support.

When we change the DE spacing to linear, specifically -90 to +90 by 1, the wavefront produces all of the eigenrays that we would expect. There is a caveat to this as well, the time step of the wavefront cannot be too large or once again we experience shadow, or missing, eigenrays. Tests showed that for a time step of 0.1 many of the eigenrays were shadowed, where as with a time step of 0.01 and smaller, no eigenrays were shadowed.

Tibonium, 2015-04-27:
We came up with an idea where in the monostatic scenarios, the wavefront DE fan can be augmented to include extra rays in the vertical directions. This would resolve this issue and not prevent any loss of data for any bistatic eigenrays that needed to be produced.

campreilly, 2015-05-06:
Updated the title to reflect the work that still needs to be done.  We may want to explore the option of using the GRAB ray fan scheme for this augmentation.

PR #204 (2015-05-13): Augmented seq_vectors at vector upper/lower bounds
Resolves #141 by adding a new seq_vector class called seq_augment that augments the passed in seq_vector to add new entries at the upper and lower limits of the container. This new class is of particular use to monostatic scenarios, where eigenrays are required to be produced by the source as its own target.
closed 2015-05-13 by islanderjtn