campreilly/UnderSeaModelingLibrary #143: Update wave_queue::add_eigenverb_listener() and add_eigenray_listener()

campreilly, 2015-04-09:
- Taking ownership of the eigenverb_listener is a problem because the collection is going to live longer than the wave_queue.
- It would be tidier if both of them looked like they were using the same design pattern, with an abstract interface and multiple listeners.
- add_eigenray_listener is using an std::list when a std::set would be much faster.
- Using BOOST_FOREACH in notify_eigenray_listeners would make it much clearer.
- I don't think that any of these add/remove/notify routines need to return a status.  How are we using this to respond to failures?

campreilly, 2015-04-10:
I've been thinking of ways to move things out of wave_queue.cc because 1200+ lines seems like a bit much.  What if we moved the "listener" servicing routines to something like "wave_producer" and let "wave_queue" inherit from that?  Not really a big change, but if we are mucking with those parts anyway, might be worth it.

Tibonium, 2015-04-27:
- wave_queue does not take ownership of the eigenverb_listener. This changed a long time ago.
- eigenverbs are sensor dependent not pair dependent, as such I at present cannot see why anyone else needs to listen for the collection. With the current design pattern, the eigenverbs go back to the sensor anyways, as does the eigenrays, which doesn't take advantage of our multiple listener pattern. If we could pass the pairs to be listeners for the eigenverbs/eigenrays then I could see this being a good addition, but seeing as eigenverbs can be created without any pairs in existence, the multiple listener pattern fails.
- are you thinking of the container in sensor_model that stores the listeners??? wave_queue has always have a vector that stores its listeners. I'm really confused...
- the return status was a design pattern choice, these return values at present are not used for failure response.

Tibonium, 2015-04-27:
In response to the code from wave_queue being large, perhaps breaking the code into two pieces would be best. One piece for propagation mathematics and another for building and cataloging. Perhaps wave_queue_propagation and wave_queue_processor? Just ideas

campreilly, 2015-05-06:
After talking about this for a while, I think we have a way forward.  What I'd like to see happen is for us to migrate wave_queue eigenray and eigenverb listeners to a structure closer to envelope_listener/envelope_notifier.

islanderjtn, 2015-05-07:
Resolved via Pull Request #191.

closed 2015-05-07 by islanderjtn