campreilly/UnderSeaModelingLibrary #145: Usml_test suite error

Tibonium, 2015-04-10:
When the usml_test suite is run, a crash occurs in eigenray_test/eigenray_basic stating that "NetCDF: Variable not found", this error goes away if the eigenray_test suite is run on its own. However, in conjunction with all the other tests, it crashes. This error does not appear to exist in the current master branch.

campreilly, 2015-04-10:
When running in Windows using Debug.  Linux seems fine.

campreilly, 2015-04-10:
Were you seeing errors like this?

```
=== reflection_test: reflect_grid_test ===
load bathymetry
writting tables to /home/sreilly/Projects/usml/waveq3d/test/reflect_grid_test.csv
time step = 0.1 secs
Check failed in file /usr/local/distrib/boost_1_57_0/boost/numeric/ublas/storage.hpp at line 195:
i < size_
unknown location(0): fatal error in "reflect_grid_test": std::out_of_range: bad index
/home/sreilly/Projects/usml/waveq3d/test/eigenray_test.cc(579): last checkpoint
```

Note that this error may be unique to the eigenverb_width branch.

campreilly, 2015-04-10:
False alarm - The reflection_test: reflect_grid_test error seems to just be a problem on my eigenverb_width branch.

islanderjtn, 2015-04-14:
Added static method to thread_task::num_active. The num_active method will return the number of active tasks in the thread_pool.
This was required  so each test in the usml_test did not interfere with subsequent test. 
The num_active method in a while loop effectively waits till all tasks have completed when in the thread_pool.
Added static reset method to all singletons that sets the _instance member to empty or NULL.
This resolves #145 in which the usml_test was crashing.

PR #150 (2015-04-16): Added static method to thread_task::num_active. The num_active method wi...
...ll return the number of active tasks in the thread_pool.

The num_active method in a while loop effectively waits till all tasks have completed when in the thread_pool.
Added static reset method to all singletons that sets the _instance member to empty or NULL.
This resolves #145 in which the usml_test was crashing.
commit 9fe50df0 (2015-04-16): Added static method to thread_task::num_active. The num_active method will return the number of active tasks in the thread_pool.
The num_active method in a while loop effectively waits till all tasks have completed when in the thread_pool.
Added static reset method to all singletons that sets the _instance member to empty or NULL.
This resolves #145 in which the usml_test was crashing.
closed 2015-04-14 by islanderjtn