campreilly/UnderSeaModelingLibrary #146: Check the accuracy of collide_from_below()

campreilly, 2015-04-10:
This method has a comment it in that says, "need to rectify this code, it mimics the bottom_reflection but signs should be changed to adjust for approaching from below."

We may be able to simplify both collide_from_below() and collide_from_above() if we assume that the volume scattering layer is nearly flat so that a modified version of surface reflection can be used.

Tibonium, 2015-04-16:
This is in reference to the computation of the grazing angle. While the grazing angle is always calculated for an interaction with the bottom, the same is not done for interactions with the surface.

To allow volume collisions to produce full eigenverbs, the grazing angle is needed when colliding from below a volume layer. I simply copied the logic for computing grazing angle in the bottom reflection case into this function and then added a note to verify that I did not have a sign error in the future. (Volumes have not yet been set to be tested and so the importance of this note was lower than current issues)

campreilly, 2015-04-16:
I think this logic can be greatly simplified if we assume that the volume layers are pretty darn flat and adapt the surface reflection code for reflection direction.

commit bbf372bb (2015-04-27): Resolves #146
    - Under the assumption that the volume layers are flat or almost flat, the logic
      for computing the grazing angle for these layers has been moved to the detect_volume_reflections
      call and greatly simplified.
    - Removed a depricated member variable, is_ray_valid, that was missed when this was removed
      from wave_queue.
    - Added another typedef to sensor_pair_manager that corresponds to the type of item stored in the
      sensor_query_map.
closed 2015-04-27 by Tibonium