campreilly/UnderSeaModelingLibrary #152: Logic error in sensor_pair_manager for multistatic sources

campreilly, 2015-04-15:
The logic for adding and removing sensors **seems to** miss the case where a multistatic capable source needs to be paired with multistatic receivers.  The current logic in the sensor_pair_manager::add_sensor() looks like this:

```
	// add monostatic pair for sensors when mode=BOTH

	if (sensor->mode() == usml::sensors::BOTH) {
		add_monostatic_pair(sensor);

	// add multistatic pairs when multistatic is true

	} else {
		if (sensor->mode() == usml::sensors::SOURCE) {
			if (sensor->source()->multistatic()) {
				add_multistatic_source(sensor);
			}
		} else {
			if (sensor->receiver()->multistatic()) {
				add_multistatic_receiver(sensor);
			}
		}
	}
```

Perhaps this needs to change to

```
	// add monostatic pair for sensors when mode=BOTH

	if (sensor->mode() == usml::sensors::BOTH) {
		add_monostatic_pair(sensor);
        }

	// add multistatic pairs when multistatic is true

	if(sensor->source()->multistatic()) {
	    switch (sensor->mode()) {
	    case usml::sensors::SOURCE:
		    add_multistatic_source(sensor);
		    break;
	    case usml::sensors::RECEIVER:
		    add_multistatic_receiver(sensor);
		    break;
	    case usml::sensors::BOTH:
		    add_multistatic_source(sensor);
		    add_multistatic_receiver(sensor);
		    break;
	    default:
		    break;
	    }
   }
```
Which looks more like the block where we add sensorID to the lists of active sources and receivers in the code just above this block.

Note that sensor_pair_manager::remove_sensor() needs something similar.

campreilly, 2015-04-16:
Assumption error: The reverberation model is used to model mutual interference between sensors.  Every source / receiver pair that has overlapping frequency bands needs to produce both direct blast returns (called fathometers in the design doc) and reverberation backscattering envelopes so that the sonar training system can model mutual interference.  Sources may include sensors associated with acoustic contacts.  These sources will be passed to the reverberation model as active sonar assets.

islanderjtn, 2015-04-16:
When adding sensor_pairs ensure that the frequency range of the source and receiver overlap. Do not add a multistatic pair when the frequency ranges do not overlap.

islanderjtn, 2015-04-16:
Done. Committed to master, commit number 83c0681e40cc3135d15ede60183e3fdcf98eb651

commit 76d44024 (2015-04-16): Moved frequencies from source_params to sensor_params for frequencies in both source_params and receiver_params.
Moved multistatic flag of sensor_params, source_params, and receiver_params to last argument of constructors, setting to true as default and allowing for optional use. Updated test's and studies with new constructor use.
Resolved #152 logic error in add/remove sensor_pair of sensor_pair_manager. Use sensor_manager_test to verify.
Previously resolved #142 memory leak error.
closed 2015-04-16 by islanderjtn