campreilly/UnderSeaModelingLibrary #156: Develop a beam_pattern_model that combines different horizontal and vertical patterns.

campreilly, 2015-04-20:
Some test cases use a beam pattern that is a combination of different types in the horizontal and vertical direction.  For example, it might use a For example a beam_pattern_grid in the vertical but a beam_pattern_solid in the horizontal.

Tibonium, 2015-04-20:
Is this any different them simply multiplying two beam patterns together?

The information contained in this message, and any attachments, may contain privileged and/or proprietary information that is intended solely for the person or entity to which it is addressed. Moreover, it may contain export restricted technical data controlled by Export Administration Regulations (EAR) or the International Traffic in Arms Regulations (ITAR). Any review, retransmission, dissemination, or re-export to foreign or domestic entities by anyone other than the intended recipient in accordance with EAR and/or ITAR regulations is prohibited.

campreilly, 2015-04-20:
Yes, the patterns should just be multiplied together.

Tibonium, 2015-04-21:
Tests have yet to be added for these new classes.

closed 2015-04-27 by islanderjtn