campreilly/UnderSeaModelingLibrary #159: Propagation Frequencies

Tibonium, 2015-04-21:
When a sensor receives an update it then calls run_wave_generator() and within this call it creates a wavefront and propagates it. At this time, the sensor passes in the _source->frequencies() as the wavefront frequencies. What happens for the following:

1) _source is not defined, ie the sensor is a receiver only
2) when the sensor is a receiver only what are the correct frequencies to use for propagation?

campreilly, 2015-04-21:
That is an excellent question; clearly something that we missed.

I was confused at first because I forgot that _source is a reference to source_params and that _receiver is a reference to _receiver_params.

campreilly, 2015-04-22:
After much discussion this morning, we decided on the following course of action:

- Change ```_frequencies( freq.clone()``` to ```_frequencies( freq )``` in the ```wave_queue``` constructor.
- Remove ```delete _frequencies ;``` from the ```wave_queue``` destructor.
- Warn the user in the doxygen header for the wave_queue constructor that the ```freq``` argument must exist for the duration of the wave_queue run.  For example, using a location variable to declare the frequenices in wavefront_generator constructor would cause a problem when that local variable went out of scope.

campreilly, 2015-04-24:
Re-reading the EVA spec indicates that 
- The frequencies for the sensor_pair should be those source frequencies that are inside the frequency band of the receiver.
- These are the frequencies used to return reverberation envelopes and fathometers to the sonar training system.  
- There is no requirement to interpolate  reverberation envelopes and fathometers to frequencies other than those defined when the sensor_pair was created.
- The EVA specification includes the ability to support multiple source beams.  I made a mistake and we'll need to fix that.

Tibonium, 2015-04-24:
So should the "frequencies" part of the wavefront be the intersection of every frequency band of the "targets" and the "source"? And then when the eigenrays are passed to the pair, we cherry-pick the intensities and phases that are valid for this pair?

campreilly, 2015-04-24:
The wavefront will only knows the frequencies for one sensor.  The sensor_pair starts with a copy of the source frequencies, but then deletes frequencies that are out of the receiver ban.  That trimming should effect the sensor_pair's fathometers and envelopes, but not the sensor_model's eigenrays and eigenverbs.

Tibonium, 2015-04-27:
I believe we have found out that the above comment is actually in reverse? The source frequencies are clipped to the receiver?

campreilly, 2015-04-27:
Based on the emails with John P. today, the source frequencies should be clipped to the receiver.

islanderjtn, 2015-04-29:
Pull Request #173 Relocate Frequencies closes all the following:
* Added seq_linear frequencies property to sensor_model manipulated by min/max active frequencies.
* Added beam_list to source_params.
* Changed sensor_pair to hold fathometer_model vs eigenrays.
* Added std::vector frequencies property to sensor_pair based on the intersection of receiver frequencies and source frequencies.
* Added to sensor_manager_test to verify changes.
Opened new issue #174 Interpolate Frequencies for items that were not addresses by Pull Request #173.

closed 2015-04-29 by islanderjtn