campreilly/UnderSeaModelingLibrary #160: Pointers and seq_vector

Tibonium, 2015-04-21:
In eigenverb and eigenray structures a pointer to a const seq_vector is stored, which corresponds to the frequencies used during the wavefront propagation. What happens to these pointers when the wave_queue destructor is called? All of the pointers in every single structures are invalidated, because wave_queue destructor deletes the pointer.

As an ad hoc at this time, we have changed a line in build_eigenray to store a clone of the frequencies in the wavefront, this resolves the problem we were previously having but now a proper solution is needed.

islanderjtn, 2015-04-22:
Closed in write_fathometers pull request.

campreilly, 2015-04-22:
Here are my notes from this morning discussion of a long term solution.

- Move the frequencies property from the sensor_params class to the sensor_model class.
  - Use this as the frequency for generating eigenrays and eigenverbs in wavefront_generator.
  - The sensor_model object is responsible for creating and destroying the memory for its frequencies property.  The wave_queue will no longer destroy this memory on behalf of the caller.
  - Use a this as the frequency axis when creating beam_pattern_grid objects for this asset.
  - Use this as the frequency for determining frequency overlap when creating sensor_pair objects.  Assume that the first frequency in the sensor is the low end of the spectrum and that the last frequency is the high end of the spectrum.
  - When creating sensors from EVAR assets, manipulate the frequencies property such that the band is limited by [MinActiveFreq,MaxActiveFreq].  If [MinActiveFreq,MaxActiveFreq] is larger than the the define frequency band, the band is extended to cover [MinActiveFreq,MaxActiveFreq].
  - Avoid using seq_data for this frequency property.  The seq_linear and seq_log perform interpolation much faster.  Adding additional smarts at creation time could save a lot of computing in highly multistatic scenarios.
- Add a frequency property to sensor_pair.
  - At the time that the sensor_pair is created, define its frequency property as the intersection of the source and receiver frequency properties.
  - The sensor_model object is responsible for creating and destroying the memory for its frequencies property.  The wave_queue will no longer destroy this memory on behalf of the caller.
  - Use this as the frequency for generating envelopes in envelope_generator.  
  - During envelope generation, the source and receiver's eigenverbs will need to be interpolated onto the sensor_pair's list of frequencies.
  - When fathometers are received from either sensor_model, they will need to be interpolated onto the sensor_pair's list of frequencies.
  - When a request for reverberation fathometers or eigenverbs comes in from the sonar_training_system, the values stored in the sensor_pair will need to be interpolated onto the frequency axis defined in the request.  This axis may not be the same as the one stored in the sensor_pair.
  - Avoid using seq_data for this frequency property.  The seq_linear and seq_log perform interpolation much faster.  Adding additional smarts at creation time could save a lot of computing in highly multistatic scenarios.
- Add the ability to cache the beam level, as a function of frequency, for each eigenverb.
  - Using the frequencies stored in each sensor_model.
  - This prevents us from recomputing beam levels in highly multistatic scenarios.

PR #161 (2015-04-22): Write fathometers
Resolved #160 . Added write_fathometers method to write fathometers to netCDF file.
commit 63d58d23 (2015-04-22): Resolved #160 issue, Pointers and seq_vector. Fix is that a clone of frequencies is not created internally in wave_queue and wave_queue destructor does not delete the cloned pointer to frequencies. The caller of wave_queue is now responsible for the deletion of the seq_vector* to frequencies passed in to wave_queue constructor.
closed 2015-04-22 by Tibonium