campreilly/UnderSeaModelingLibrary #168: Of Frequencies and Eigenverbs

Tibonium, 2015-04-27:
Eigenverbs are a frequency dependent entity and are specific to a sensor. If this is so, there is a major concern that has yet to be addressed when computing envelope contributions.

That is the frequencies for one sensor is not guaranteed to be either the same number or even the same frequency band compared to another. This is an issue, as the convolution process requires that both eigenverbs be the same frequency band and hence the frequencies vector are equivalent in size and values. This implies that the widths/lengths and energies are all valid items to be convolved together.

campreilly, 2015-04-27:
I am currently working to address that issue on the envelope_width branch.  

I moved the core envelope calculation into a class called envelope_model.  It assumes that the source and receiver eigenverb have already been interpolated onto a common frequency axis, then compute the envelope contribution as a function of that frequency and time.  It passes that result to envelope_collection, which applies the various combinations of source and receiver beam pattern, and then stores the result.

For example, if we had a source that operates at 1000, 2000, 3000, 4000 Hz, and a receiver that operates at 500, 1200 1800, 2500 Hz, the frequencies for the sensor_pair would current be just 1000 and 2000 Hz.  Each receiver eigenverb would need to be interpolated onto these frequencies, but the source eigenverb would just need to be sub-setted, which should be faster.

campreilly, 2015-04-27:
Related to issue #159

campreilly, 2015-05-05:
Resolved with the changes implemented in pull request #181 .

closed 2015-05-05 by campreilly