campreilly/UnderSeaModelingLibrary #174: Interpolate frequenices

islanderjtn, 2015-04-29:
From #159 Propagate Frequencies the items that were not closed by Pull Request #173 .

Using the frequency property of sensor_pair:

* Use this frequency for generating envelopes in envelope_generator.
During envelope generation, the source and receiver's eigenverbs will need to be interpolated onto the sensor_pair's list of frequencies.
* When fathometers are received from the sensor_model, they will need to be interpolated onto the sensor_pair's list of frequencies. We'll need to be very clever to understand how to optimize this interpolation.
* When a request for reverberation fathometers or eigenverbs comes in from the sonar_training_system, the values stored in the sensor_pair will need to be interpolated onto the frequency axis defined in the request. This axis may not be the same as the one stored in the sensor_pair. We'll need to be very clever to understand how to optimize this interpolation.

Add the ability to cache the beam level, as a function of frequency, for each eigenverb.
* Using the frequencies stored in each sensor_model.
* This prevents us from recomputing beam levels in highly multistatic scenarios.

campreilly, 2015-04-30:
Is this the same as issue #172?

campreilly, 2015-05-06:
Most of this work was completed as part of the relocate_frequencies branch. Parts that remain were split into their own issues.

closed 2015-05-06 by campreilly