campreilly/UnderSeaModelingLibrary #177: Release 1.0 fails to compile in pedantic mode

campreilly, 2015-05-04:
The reflection_test.cc has an unitialized variable.

```
[ 98%] Building CXX object CMakeFiles/usml_test.dir/waveq3d/test/reflection_test.cc.o
cc1plus: warnings being treated as errors
/home/sreilly/Projects/usml/waveq3d/test/reflection_test.cc: In member function ‘void reflection_test::reflect_flat_test::test_method()’:
/home/sreilly/Projects/usml/waveq3d/test/reflection_test.cc:151: error: ‘callback.reflection_test::reflection_callback::latitude’ may be used uninitialized in this function
make[2]: *** [CMakeFiles/usml_test.dir/waveq3d/test/reflection_test.cc.o] Error 1
make[1]: *** [CMakeFiles/usml_test.dir/all] Error 2
make: *** [all] Error 2
```

Tibonium, 2015-05-04:
https://github.com/campreilly/UnderSeaModelingLibrary/commit/c48adbdfde5da14e0a863b16617839b1cd4bc011

Release1.0????

campreilly, 2015-05-04:
Yes, release 1.0.  I've now checked it twice on my machine.

Tibonium, 2015-05-04:
I can't see your machine, but whatever you have for release1.0 it is NOT what is on this site. Odds are you did an accidental pull from master into your local branch of release1.0, which would explain why you have the master changes to relfection_test and it is not on the site. Odds are you would also see that if you did a "git status" it would say that your "branch is ahead of origin by x commits" and you have unpushed commits.

https://github.com/campreilly/UnderSeaModelingLibrary/blob/release1.0/waveq3d/test/reflection_test.cc

campreilly, 2015-05-04:
You are correct.  The mistake is on my side.  Correcting now.

closed 2015-05-04 by campreilly