campreilly/UnderSeaModelingLibrary #178: interp_speed_test not showing speed advantage for fast interpolation

campreilly, 2015-05-04:
When I run datagrid_test/interp_speed_test, the fast and slow methods seem to run at almost the same speed.  Sometimes, the fast_grid is even a little bit behind.

* Increased the number of points from 1e3 to 1e6.
* Cent O/S in Release mode.
* Release 1.0.0 version on release 1.0 branch

```
[sreilly@sreilly usml_release1.0.0]$ usml_test --run_test=datagrid_test/interp_speed_test
Running 1 test case...
=== datagrid_test: interpolation_speed_test ===
Interpolation using data_grid method
0.35 s

Interpolation using fast_grid method
0.44 s
```

campreilly, 2015-05-04:
Also reported by John Pinezich at AAC.

```
From: Pinezich, John [mailto:JPinezich@aaccorp.com] 
Sent: Monday, May 04, 2015 8:53 AM
To: Sean Reilly
Subject: interp_speed_test

Sean, 

When I do datagrid_test/interp_speed_test, the fast_grid  method is quite a bit slower than the grid method.  I changed number of points to 1e4, and 1e5, and see about 6:1 speed ratio.

![capture](https://cloud.githubusercontent.com/assets/1370874/7454429/9474b4d2-f241-11e4-9d9b-cae58d562d96.png)

John
```

Tibonium, 2015-05-04:
For small numbers we do not see the advantage, but rather for large, really large numbers. If you were to run for 1e7 points and higher on my machine I begin to see faster responses. As the points increase the speed discrepancy becomes more and more prevalent. I showed a 10% increase in speed for 1e8.

campreilly, 2015-05-04:
When I increase the number of items to 1e8 from 1e3, I see the speed advantage too.

closed 2015-05-04 by campreilly