campreilly/UnderSeaModelingLibrary #183: Fathometer calculation by wavefront_generator running slower than expected.

campreilly, 2015-05-06:
Execution of reverb_analytic_test with just the fathometer calculation is taking up to 3 seconds.  This might be nothing, but we need to investigate.

campreilly, 2015-05-06:
Here is an example that took 4.92 sec on Sean's machine.

```
[sreilly@sreilly usml]$ ./reverb_analytic_test
== define ocean characteristics ==
== define sensor characteristics ==
== deploy sensor instance ==
******
sensor_manager: add sensor(1)
sensor_pair_manager: add sensor(1)
   add_monostatic_pair: sensor_pair(1,1)
sensor_pair_manager:  current pairs
     pair  src_rcv 1_1
******
sensor_manager: update sensor(1)
sensor_model: update_sensor(1)
sensor_model: run_wave_generator(1)
== wait for results ==
sensor_model: update_eigenrays(1)
sensor_pair: update_fathometer(1)
waited for 4.92 secs
== test complete ==
```

campreilly, 2015-05-06:
This may be a red herring.  The same test, run under Linux in Release mode only took 0.25 sec.
```
== define ocean characteristics ==
== define sensor characteristics ==
== deploy sensor instance ==
== wait for results ==
waited for 0.25 secs
== test complete ==
```

islanderjtn, 2015-05-07:
It is a red herring. My results on Linux in Release mode only took 0.2 sec.

./reverb_analytic_test
== define ocean characteristics ==
== define sensor characteristics ==
== deploy sensor instance ==
== wait for results ==
waited for 0.2 secs
== test complete ==

closed 2015-05-07 by islanderjtn