campreilly/UnderSeaModelingLibrary #190: Fix doxygen errors in sensor and eigenverb modules

campreilly, 2015-05-06:
Running doxygen exposes a host of issues
```
[sreilly@sreilly usml]$ doxygen
/home/sreilly/Projects/usml/sensors/receiver_params.cc:16: Warning: no matching class member found for
  receiver_params(int paramsID, double min_freq, double max_freq, const usml::types::seq_vector &frequencies, const std::list< int > &beamList, bool multistatic)
Possible candidates:
  receiver_params::receiver_params(sensor_params::id_type paramsID, double min_freq, double max_freq, const seq_vector &frequencies, const beam_pattern_list &beam_list, bool multistatic=true)
  receiver_params::receiver_params(const receiver_params &other)
/home/sreilly/Projects/usml/ocean/test/ocean_shared_test.cc:4: Warning: included file threads/test/ocean_shared_test.cc is not found. Check your EXAMPLE_PATH
/home/sreilly/Projects/usml/sensors/source_params.h:78: Warning: Found unknown command `\1m'
/home/sreilly/Projects/usml/sensors/test/maps_test.cc:23: Warning: Found unknown command `\updated'
/home/sreilly/Projects/usml/eigenverb/eigenverb_collection.h:39: Warning: argument 'num_volume' of command @param is not found in the argument list of eigenverb_collection::eigenverb_collection(size_t num_volumes)
/home/sreilly/Projects/usml/eigenverb/eigenverb_collection.h:39: Warning: The following parameters of usml::eigenverb::eigenverb_collection::eigenverb_collection(size_t num_volumes) are not documented:
  parameter 'num_volumes'
/home/sreilly/Projects/usml/eigenverb/eigenverb_collection.h:76: Warning: The following parameters of usml::eigenverb::eigenverb_collection::add_eigenverb(const eigenverb &verb, size_t interface) are not documented:
  parameter 'verb'
/home/sreilly/Projects/usml/eigenverb/eigenverb_collection.h:172: Warning: end of comment block while expecting command </pre>
/home/sreilly/Projects/usml/eigenverb/eigenverb_collection.h:172: Warning: end of comment block while expecting command </pre>
/home/sreilly/Projects/usml/eigenverb/envelope_generator.h:33: Warning: argument 'curves' of command @param is not found in the argument list of envelope_generator::generate_envelopes(const eigenverb_collection &source, const eigenverb_collection &receiver, envelope_collection *levels)
/home/sreilly/Projects/usml/eigenverb/envelope_generator.h:33: Warning: The following parameters of usml::eigenverb::envelope_generator::generate_envelopes(const eigenverb_collection &source, const eigenverb_collection &receiver, envelope_collection *levels) are not documented:
  parameter 'levels'
/home/sreilly/Projects/usml/sensors/beam_pattern_solid.cc:11: Warning: Found unknown command `\elem'
/home/sreilly/Projects/usml/sensors/beam_pattern_solid.cc:11: Warning: Found unknown command `\elem'
/home/sreilly/Projects/usml/sensors/fathometer_model.h:37: Warning: argument 'eigenrays' of command @param is not found in the argument list of fathometer_model::fathometer_model(sensor_model::id_type source_id, sensor_model::id_type receiver_id, wposition1 src_pos, wposition1 rcv_pos, const eigenray_list &list)
/home/sreilly/Projects/usml/sensors/fathometer_model.h:37: Warning: The following parameters of usml::sensors::fathometer_model::fathometer_model(sensor_model::id_type source_id, sensor_model::id_type receiver_id, wposition1 src_pos, wposition1 rcv_pos, const eigenray_list &list) are not documented:
  parameter 'list'
/home/sreilly/Projects/usml/sensors/fathometer_model.h:124: Warning: argument 'The' of command @param is not found in the argument list of fathometer_model::depth_offset(double depth_offset)
/home/sreilly/Projects/usml/sensors/fathometer_model.h:124: Warning: The following parameters of usml::sensors::fathometer_model::depth_offset(double depth_offset) are not documented:
  parameter 'depth_offset'
/home/sreilly/Projects/usml/sensors/fathometer_model.h:108: Warning: argument 'The' of command @param is not found in the argument list of fathometer_model::distance_from_sensor(double distance_from_sensor)
/home/sreilly/Projects/usml/sensors/fathometer_model.h:108: Warning: The following parameters of usml::sensors::fathometer_model::distance_from_sensor(double distance_from_sensor) are not documented:
  parameter 'distance_from_sensor'
/home/sreilly/Projects/usml/sensors/fathometer_model.h:181: Warning: argument 'eigenray_list' of command @param is not found in the argument list of fathometer_model::eigenrays(eigenray_list list)
/home/sreilly/Projects/usml/sensors/fathometer_model.h:181: Warning: The following parameters of usml::sensors::fathometer_model::eigenrays(eigenray_list list) are not documented:
  parameter 'list'
/home/sreilly/Projects/usml/sensors/fathometer_model.h:84: Warning: argument 'The' of command @param is not found in the argument list of fathometer_model::receiver_id(sensor_model::id_type receiver_id)
/home/sreilly/Projects/usml/sensors/fathometer_model.h:84: Warning: The following parameters of usml::sensors::fathometer_model::receiver_id(sensor_model::id_type receiver_id) are not documented:
  parameter 'receiver_id'
/home/sreilly/Projects/usml/sensors/fathometer_model.h:164: Warning: argument 'The' of command @param is not found in the argument list of fathometer_model::receiver_position(wposition1 position)
/home/sreilly/Projects/usml/sensors/fathometer_model.h:164: Warning: The following parameters of usml::sensors::fathometer_model::receiver_position(wposition1 position) are not documented:
  parameter 'position'
/home/sreilly/Projects/usml/sensors/fathometer_model.h:92: Warning: argument 'The' of command @param is not found in the argument list of fathometer_model::slant_range(double slant_range)
/home/sreilly/Projects/usml/sensors/fathometer_model.h:92: Warning: The following parameters of usml::sensors::fathometer_model::slant_range(double slant_range) are not documented:
  parameter 'slant_range'
/home/sreilly/Projects/usml/sensors/fathometer_model.h:68: Warning: argument 'The' of command @param is not found in the argument list of fathometer_model::source_id(sensor_model::id_type source_id)
/home/sreilly/Projects/usml/sensors/fathometer_model.h:68: Warning: The following parameters of usml::sensors::fathometer_model::source_id(sensor_model::id_type source_id) are not documented:
  parameter 'source_id'
/home/sreilly/Projects/usml/sensors/fathometer_model.h:148: Warning: argument 'The' of command @param is not found in the argument list of fathometer_model::source_position(wposition1 position)
/home/sreilly/Projects/usml/sensors/fathometer_model.h:148: Warning: The following parameters of usml::sensors::fathometer_model::source_position(wposition1 position) are not documented:
  parameter 'position'
/home/sreilly/Projects/usml/sensors/orientation.cc:19: Warning: The following parameters of usml::sensors::orientation::orientation(double heading, double pitch, double roll, vector< double > ref_axis) are not documented:
  parameter 'ref_axis'
/home/sreilly/Projects/usml/sensors/sensor_listener.h:43: Warning: argument 'eigenray_list' of command @param is not found in the argument list of sensor_listener::update_fathometer(int sensorID, eigenray_list *list)
/home/sreilly/Projects/usml/sensors/sensor_listener.h:43: Warning: The following parameters of usml::sensors::sensor_listener::update_fathometer(int sensorID, eigenray_list *list) are not documented:
  parameter 'list'
/home/sreilly/Projects/usml/sensors/sensor_manager.h:35: Warning: Found unknown command `\updated'
/home/sreilly/Projects/usml/sensors/sensor_model.h:161: Warning: argument 'shared_pointer' of command @param is not found in the argument list of sensor_model::eigenverbs() const
/home/sreilly/Projects/usml/sensors/sensor_model.cc:220: Warning: Found unknown command `\params'
/home/sreilly/Projects/usml/sensors/sensor_model.cc:233: Warning: Found unknown command `\params'
/home/sreilly/Projects/usml/sensors/sensor_model.cc:83: Warning: argument 'orientation' of command @param is not found in the argument list of sensor_model::update_sensor(const wposition1 &position, const orientation &orient, bool force_update=false)
/home/sreilly/Projects/usml/sensors/sensor_model.cc:83: Warning: The following parameters of usml::sensors::sensor_model::update_sensor(const wposition1 &position, const orientation &orient, bool force_update=false) are not documented:
  parameter 'orient'
/home/sreilly/Projects/usml/sensors/sensor_pair.cc:28: Warning: argument 'eigenray_list' of command @param is not found in the argument list of sensor_pair::update_fathometer(sensor_model::id_type sensorID, eigenray_list *lis [... cut ...]
 french, greek, monogreek, ancientgreek, croatian, hungarian, interlingua, ibyc
us, indonesian, icelandic, italian, latin, mongolian, dutch, norsk, polish, por
tuguese, pinyin, romanian, russian, slovenian, uppersorbian, serbian, swedish, 
turkish, ukenglish, ukrainian, loaded.
```
There are also many places where:
- @param statements have variable name but no description
- File header is missing the "brief description" used in the File List.
- Module names are messed up, especially on sensors and beam patterns.

[image]

Tibonium, 2015-05-08:
"Module names are messed up, especially on sensors and beam patterns."
Then what are they?

campreilly, 2015-05-08:
In all of the other modules, the directory off usml is synonymous with the module.  The sensors module does not follow that pattern.

Tibonium, 2015-05-08:
````
nclude <usml/types/wvector.h>
#include <usml/types/seq_vector.h>

using namespace usml::ublas;

namespace usml {
namespace types {

/// @ingroup data_grid
/// @{

    /** Type of interpolation used for each axis. */
    enum GRID_INTERP_TYPE
    {
        GRID_INTERP_NEAREST = -1, GRID_INTERP_LINEAR = 0, // default
        GRID_INTERP_PCHIP = 1
    };
````

islanderjtn, 2015-05-11:
Resolved by #202

PR #202 (2015-05-11): Doxygen cleanup
Resolves #190.
Added File List descriptor for all files that were missing one or had an issue preventing it showing up in the html.
Went through all other errors presented by doxygen and resolved them. Looked for, and resolved, empty @param 's.
closed 2015-05-11 by islanderjtn