campreilly/UnderSeaModelingLibrary #201: Test the accuracy of fathometers and eigenverb produced by wave_generator

campreilly, 2015-05-11:
Currently, we rely on the reverb_analytic_test to store fathometer and eigenverb results in netCDF files.  But, we only test that the netCDF files have data in them (see #199); we should also test that the results are accurate.  For eigenverb, we should be able to use some of the techniques in eigenverb_test.cc to create a set of Matlab routines that read the netCDF file and tests the accuracy of the results.  We'll need to build an analytic solution for the fathometers.

At the same time, we should update reverb_analytic_test to ensure that the parameters still match those in the paper's analytic solution.  I noticed that it no longer using a frequency of 3000 Hz.

campreilly, 2015-05-19:
My intent is to have us build some Matlab routines that load the fathometer and eigenverb results from the netCDF files, and then compare their values to analytic solutions in Matlab.

Tibonium, 2015-05-27:
I believe it is possible to lay the ground work for this issue for the eigenverbs, as it still depends on #215 . The fathometer portion however can be completed and compared with analytic solutions. In the 200m water with constant sound speed, analytic solutions exist.

Tibonium, 2015-05-27:
At present there is a lot of interim work going on from the last pull that was committed to master. After doing so there are many bugs that have been introduced into the reverb_analytic_test. Fathometers are not being produced and therefore cannot be inspected. And since eigenverbs have yet to be updated with the resolution of #215, I'm going to put this issue on hold until it is resolved.

campreilly, 2024-01-05:
Specifically, look at sensors_test/update_wavefront_data and biverbs_test/update_wavefront_data.
Compare acoustic paths to analytic solutions to validate number of paths for each pair.