campreilly/UnderSeaModelingLibrary #205: Unable to repeat sequences of random numbers using randgen::seed(0);

campreilly, 2015-05-12:
From: Pinezich, John [mailto:JPinezich@aaccorp.com] 
Sent: Tuesday, May 12, 2015 12:12 PM
To: Sean Reilly
Subject: RE: randgen

That is what I tried to do, but it didn't work:
```
    int num_points = 10;
    if (num_points > 1)
       randgen::seed(0);
    for(int i=0; i<num_points; ++i)
    {
       double spot[2] ;
       double result;

       spot[0] = 39;
       spot[1] = -49;

       if (i>0)
          oBathy.getRandomLocation(spot,GEODETIC_COORDS);

       printf("%2d %10.5f %10.5f degs  ",i,spot[0],spot[1]);
       spot[0] = to_colatitude(spot[0]);
       spot[1] = to_radians(spot[1]);

       double fast_result = fast_grid->interpolate( spot );
       printf("%10.5f %10.5f rads  result =  %10.5f  %10.5f  %10.5f\n",spot[0],spot[1],result,
              wposition::earth_radius - result, wposition::earth_radius - fast_result);
     }
```
and here is the call:
```
void BathyGridTest::getRandomLocation(double p_lfLocation[], CoordSystem p_eCoordSystem)
{
   p_lfLocation[0] = m_lfGridLatStart + m_lfGridSizeDegs * randgen::uniform();
   p_lfLocation[1] = m_lfGridLonStart + m_lfGridSizeDegs * randgen::uniform();
   if (p_eCoordSystem == GEODETIC_COORDS)
      return;
   p_lfLocation[0] = to_colatitude(p_lfLocation[0]);
   p_lfLocation[1] = to_radians(p_lfLocation[1]);
}
```
producing these results on two consecutive runs
```
[root@cas00 build]# ./usml_test --run_test=neo_test/neo_test_1
Running 1 test case...
=== eigenray_test: neo_test_1 ===
seed value = 0
 0   39.00000  -49.00000 degs     0.89012   -0.85521 rads  result =     0.00000  6378101.03020    25.00000
 1   44.30264  -46.23252 degs     0.79757   -0.80691 rads  result =     0.00000  6378101.03020    55.54828
 2   49.38403  -45.45535 degs     0.70888   -0.79335 rads  result =     0.00000  6378101.03020    83.88451
 3   35.60429  -40.87954 degs     0.94938   -0.71348 rads  result =     0.00000  6378101.03020     9.57454
 4   39.17666  -48.54336 degs     0.88703   -0.84724 rads  result =     0.00000  6378101.03020    26.19995
 5   35.11173  -54.83810 degs     0.95798   -0.95711 rads  result =     0.00000  6378101.03020     0.38740
 6   46.94252  -35.22582 degs     0.75149   -0.61481 rads  result =     0.00000  6378101.03020    75.61539
 7   41.57988  -44.83330 degs     0.84509   -0.78249 rads  result =     0.00000  6378101.03020    41.27270
 8   37.09379  -51.18095 degs     0.92339   -0.89328 rads  result =     0.00000  6378101.03020    13.42538
 9   37.89589  -35.19646 degs     0.90939   -0.61429 rads  result =     0.00000  6378101.03020    25.86912
~BathyGridTest()

*** No errors detected
[root@cas00 build]# ./usml_test --run_test=neo_test/neo_test_1
Running 1 test case...
=== eigenray_test: neo_test_1 ===
seed value = 0
 0   39.00000  -49.00000 degs     0.89012   -0.85521 rads  result =     0.00000  6378101.03020    25.00000
 1   39.34693  -36.30777 degs     0.88406   -0.63369 rads  result =     0.00000  6378101.03020    33.29342
 2   47.67944  -54.17801 degs     0.73863   -0.94558 rads  result =     0.00000  6378101.03020    70.07664
 3   42.04244  -36.15702 degs     0.83702   -0.63106 rads  result =     0.00000  6378101.03020    48.20631
 4   41.15046  -48.17677 degs     0.85259   -0.84084 rads  result =     0.00000  6378101.03020    37.23913
 5   47.73398  -42.68094 degs     0.73768   -0.74492 rads  result =     0.00000  6378101.03020    76.19642
 6   47.86980  -47.93792 degs     0.73531   -0.83667 rads  result =     0.00000  6378101.03020    74.31495
 7   44.77694  -40.92934 degs     0.78929   -0.71435 rads  result =     0.00000  6378101.03020    60.80850
 8   44.57135  -49.73657 degs     0.79288   -0.86807 rads  result =     0.00000  6378101.03020    55.27412
 9   48.34581  -51.28478 degs     0.72700   -0.89509 rads  result =     0.00000  6378101.03020    75.25955
~BathyGridTest()
```
But when I put this into randgen.cc:
```
//randgen::random_type randgen::master_gen(time(0));
randgen::random_type randgen::master_gen(0);
```
it repeats:
```
[root@cas00 build]# ./usml_test --run_test=neo_test/neo_test_1
Running 1 test case...
=== eigenray_test: neo_test_1 ===
seed value = 0
 0   39.00000  -49.00000 degs     0.89012   -0.85521 rads  result =     0.00000  6378101.03020    25.00000
 1   49.21359  -36.09517 degs     0.71186   -0.62998 rads  result =     0.00000  6378101.03020    87.68321
 2   49.58347  -39.73880 degs     0.70540   -0.69357 rads  result =     0.00000  6378101.03020    87.83967
 3   41.91432  -50.01037 degs     0.83925   -0.87285 rads  result =     0.00000  6378101.03020    40.52360
 4   54.64357  -51.68376 degs     0.61709   -0.90205 rads  result =     0.00000  6378101.03020   110.35970
 5   40.04739  -47.20069 degs     0.87184   -0.82381 rads  result =     0.00000  6378101.03020    31.66028
 6   42.46296  -48.55546 degs     0.82968   -0.84745 rads  result =     0.00000  6378101.03020    44.26855
 7   44.90954  -43.70079 degs     0.78698   -0.76272 rads  result =     0.00000  6378101.03020    60.15208
 8   44.07657  -38.38349 degs     0.80152   -0.66992 rads  result =     0.00000  6378101.03020    58.22940
 9   39.76269  -45.98432 degs     0.87681   -0.80258 rads  result =     0.00000  6378101.03020    30.70264
~BathyGridTest()

*** No errors detected
[root@cas00 build]# ./usml_test --run_test=neo_test/neo_test_1
Running 1 test case...
=== eigenray_test: neo_test_1 ===
seed value = 0
 0   39.00000  -49.00000 degs     0.89012   -0.85521 rads  result =     0.00000  6378101.03020    25.00000
 1   49.21359  -36.09517 degs     0.71186   -0.62998 rads  result =     0.00000  6378101.03020    87.68321
 2   49.58347  -39.73880 degs     0.70540   -0.69357 rads  result =     0.00000  6378101.03020    87.83967
 3   41.91432  -50.01037 degs     0.83925   -0.87285 rads  result =     0.00000  6378101.03020    40.52360
 4   54.64357  -51.68376 degs     0.61709   -0.90205 rads  result =     0.00000  6378101.03020   110.35970
 5   40.04739  -47.20069 degs     0.87184   -0.82381 rads  result =     0.00000  6378101.03020    31.66028
 6   42.46296  -48.55546 degs     0.82968   -0.84745 rads  result =     0.00000  6378101.03020    44.26855
 7   44.90954  -43.70079 degs     0.78698   -0.76272 rads  result =     0.00000  6378101.03020    60.15208
 8   44.07657  -38.38349 degs     0.80152   -0.66992 rads  result =     0.00000  6378101.03020    58.22940
 9   39.76269  -45.98432 degs     0.87681   -0.80258 rads  result =     0.00000  6378101.03020    30.70264
~BathyGridTest()

*** No errors detected
[root@cas00 build]# 
```

campreilly, 2015-05-12:
Our randgen_test has the exact same problem, and it uses randgen::seed(100).
```
[sreilly@sreilly usml]$ usml_test --run_test=randgen_test
Running 2 test cases...
=== randgen_test: randVectorTest ===
uniform:  [7](0.330409,0.435102,0.784551,0.280824,0.0489535,0.401298,0.394794)
gaussian: [7](-0.873106,1.88159,-0.603269,-0.940238,-0.171355,0.447582,1.08076)
noise:    [3]((-0.437426,0.356166),(1.17884,-0.0331394),(0.78136,0.447582))
=== randgen_test: randMatrixTest ===
uniform:  [2,3]((0.390094,0.648743,0.827554),(0.31901,0.651876,0.649617))
gaussian: [2,3]((-1.02986,0.0786533,0.628047),(0.728869,-0.239869,1.13729))
noise:    [2,3](((0.609624,-0.66879),(-0.673838,-1.32086),(-0.857083,-0.619722)),((-1.09602,-1.50213),(-0.78905,1.70019),(1.32895,2.02567e-322)))

*** No errors detected
[sreilly@sreilly usml]$ usml_test --run_test=randgen_test
Running 2 test cases...
=== randgen_test: randVectorTest ===
uniform:  [7](0.0131648,0.295966,0.926368,0.99815,0.745677,0.675958,0.062432)
gaussian: [7](-0.19595,-1.16351,1.20501,-0.665109,0.467264,-0.205897,-0.752961)
noise:    [3]((-1.45145,1.24916),(-1.07891,-1.25319),(0.775776,-0.205897))
=== randgen_test: randMatrixTest ===
uniform:  [2,3]((0.0385099,0.0884773,0.573852),(0.187267,0.0862505,0.347594))
gaussian: [2,3]((1.18633,-0.0486456,0.792545),(-0.760783,0.656815,-0.0898059))
noise:    [2,3](((0.415105,-0.356738),(-0.615465,-0.0240234),(-0.0336388,-0.933044)),((-0.410972,-0.38357),(0.305867,-0.209566),(3.23241,3.41186e-315)))

*** No errors detected
```

campreilly, 2015-05-13:
Sorry, I was a bit unclear.  When the seed is hardset, like it is in randgen_test, we should get the same set of random numbers ever time that we run the test.  But, we are getting different numbers, and that's wrong.  

If we hardset the value inside of the library itself, it does get the same number each time.  That means that it is probably something we are doing wrong rather than a problem in the library itself.

Tibonium, 2015-05-22:
Construction of and relation to the passed in parameters of the construction are the breakdown points on this issue.

It was believed that simply updating the seed value for the internal <code>master_gen</code> would be sufficient to update the other internal generators(<code>uniform_gen</code> and <code>gaussian_gen</code>). This is incorrect as the other internal generators copy the <code>master_gen</code> at construction time and hence have no relation to the internal static <code>master_gen</code>. As such when a call is made to update <code>master_gen</code>'s seed value, the other generators are not updated. The following code is needed to rectify this issue:
````
static inline void seed(random_type::result_type value) {
    uniform_gen.base().seed(value);
    gaussian_gen.engine().seed(value);
}
````

With this implementation of the function call for seeding the random numbers, the randgen_tests now produce duplicate random results between runs, as intended.

commit aa8f79b7 (2015-05-22): Resolves #205
closed 2015-05-22 by Tibonium