campreilly/UnderSeaModelingLibrary #207: Use of seq_augment in wavefront_generator causes crash

islanderjtn, 2015-05-13:
While reviewing seq_augment pull request attempted to add seq_augment to wavefront_generator creation of DE's.  When running reverb_analytic_test found following crash.
seq_rayfan use is tested in sequence_test/seq_augment_test.

[image]

campreilly, 2015-05-13:
Is it possible that some of the values in the augmented seq_rayfan are duplicates?  That would cause delta(1) to equal zero, and result in this crash.

islanderjtn, 2015-05-13:
Reviewed output values did not see duplicates.

The information contained in this message, and any attachments, may contain privileged and/or proprietary information that is intended solely for the person or entity to which it is addressed. Moreover, it may contain export restricted technical data controlled by Export Administration Regulations (EAR) or the International Traffic in Arms Regulations (ITAR). Any review, retransmission, dissemination, or re-export to foreign or domestic entities by anyone other than the intended recipient in accordance with EAR and/or ITAR regulations is prohibited.

Tibonium, 2015-05-14:
I am unable to reproduce this error, in either debug/release mode. Both complete just fine for me with your commented out code.

commit 64d822e4 (2015-05-15): Resolved #207, missed setting the second to last value of the increment.
closed 2015-05-15 by Tibonium