campreilly/UnderSeaModelingLibrary #210: Unable set to set bottom scattering_model to Lambert in reverb_analytic_test

campreilly, 2015-05-19:
The reverb_analytic_test study sets the scattering_model for the ocean bottom to scattering_lambert in its define_ocean_characteristics() method.
```
	void define_ocean_characteristics() {
		cout << "== define ocean characteristics ==" << endl;
		profile_model* water = new profile_linear(1500.0);
		water->attenuation(new attenuation_constant(0.0));
		profile_lock* profile = new profile_lock(water);

		boundary_model* surf = new boundary_flat();
		boundary_lock* surface = new boundary_lock(surf);

		boundary_model* btm = new boundary_flat(200.0);
		btm->reflect_loss(
				new reflect_loss_rayleigh(reflect_loss_rayleigh::SAND));
		btm->scattering(new scattering_lambert());
		boundary_lock* bottom = new boundary_lock(btm);

		ocean_shared::reference ocean(
				new ocean_model(surface, bottom, profile));
		ocean_shared::update(ocean);
	}
```
However, when it gets used in envelope_generator::scattering(), the model that actually gets called is scattering_constant(-300.0) which also happens to be the default for the _scattering member of boundary_model.

```
void envelope_generator::scattering( size_t interface, const wposition1& location,
		const seq_vector& frequencies, double de_incident,
		double de_scattered, double az_incident, double az_scattered,
		vector<double>* amplitude)
{
	switch ( interface ) {
	case eigenverb::BOTTOM:
		_ocean->bottom().scattering(location,frequencies,
				de_incident,de_scattered,az_incident,az_scattered,
				amplitude) ;
		break;
	case eigenverb::SURFACE:
		_ocean->surface().scattering(location,frequencies,
				de_incident,de_scattered,az_incident,az_scattered,
				amplitude) ;
		break;
	default:
		size_t layer = (size_t) floor( (interface-2.0)/2.0 ) ;
		_ocean->volume( layer ).scattering(location,frequencies,
				de_incident,de_scattered,az_incident,az_scattered,
				amplitude) ;
		break;
	}
}
```
We know that it is is scattering_constant that is getting called:

1) Because we used the debugger to step into the method.
2) The scattering strength is always returned as 1e-30, which is -300 dB in linear units.

Tibonium, 2015-05-19:
Boundary_lock has no call to scattering. This is your most likely point of failure.

The information contained in this message, and any attachments, may contain privileged and/or proprietary information that is intended solely for the person or entity to which it is addressed. Moreover, it may contain export restricted technical data controlled by Export Administration Regulations (EAR) or the International Traffic in Arms Regulations (ITAR). Any review, retransmission, dissemination, or re-export to foreign or domestic entities by anyone other than the intended recipient in accordance with EAR and/or ITAR regulations is prohibited.

islanderjtn, 2015-05-19:
Right. On it.
Ted

Tibonium, 2015-05-28:
I am unable to build and install usml on windows x64, given the following linking error.
````
1>------ Build started: Project: usml, Configuration: Release x64 ------
1>usml.dir\Release\boundary_flat.obj : fatal error LNK1136: invalid or corrupt file
2>------ Build started: Project: usml_test, Configuration: Release x64 ------
2>     Creating library C:/Users/David Thibaudeau/Documents/usml-build/Release/usml_test.lib and object C:/Users/David Thibaudeau/Documents/usml-build/Release/usml_test.exp
2>boundary_lock_test.obj : error LNK2001: unresolved external symbol "public: virtual void __cdecl usml::ocean::boundary_lock::scattering(class usml::types::wposition1 const &,class usml::types::seq_vector const &,double,double,double,double,class boost::numeric::ublas::vector<double,class boost::numeric::ublas::unbounded_array<double,class std::allocator<double> > > *)" (?scattering@boundary_lock@ocean@usml@@UEAAXAEBVwposition1@types@3@AEBVseq_vector@53@NNNNPEAV?$vector@NV?$unbounded_array@NV?$allocator@N@std@@@ublas@numeric@boost@@@ublas@numeric@boost@@@Z)
2>ocean_shared_test.obj : error LNK2001: unresolved external symbol "public: virtual void __cdecl usml::ocean::boundary_lock::scattering(class usml::types::wposition1 const &,class usml::types::seq_vector const &,double,double,double,double,class boost::numeric::ublas::vector<double,class boost::numeric::ublas::unbounded_array<double,class std::allocator<double> > > *)" (?scattering@boundary_lock@ocean@usml@@UEAAXAEBVwposition1@types@3@AEBVseq_vector@53@NNNNPEAV?$vector@NV?$unbounded_array@NV?$allocator@N@std@@@ublas@numeric@boost@@@ublas@numeric@boost@@@Z)
2>boundary_lock_test.obj : error LNK2001: unresolved external symbol "public: virtual void __cdecl usml::ocean::boundary_lock::scattering(class usml::types::wposition const &,class usml::types::seq_vector const &,double,class boost::numeric::ublas::matrix<double,struct boost::numeric::ublas::basic_row_major<unsigned __int64,__int64>,class boost::numeric::ublas::unbounded_array<double,class std::allocator<double> > >,double,class boost::numeric::ublas::matrix<double,struct boost::numeric::ublas::basic_row_major<unsigned __int64,__int64>,class boost::numeric::ublas::unbounded_array<double,class std::allocator<double> > >,class boost::numeric::ublas::matrix<class boost::numeric::ublas::vector<double,class boost::numeric::ublas::unbounded_array<double,class std::allocator<double> > >,struct boost::numeric::ublas::basic_row_major<unsigned __int64,__int64>,class boost::numeric::ublas::unbounded_array<class boost::numeric::ublas::vector<double,class boost::numeric::ublas::unbounded_array<double,class std::allocator<double> > >,class std::allocator<class boost::numeric::ublas::vector<double,class boost::numeric::ublas::unbounded_array<double,class std::allocator<double> > > > > > *)" (?scattering@boundary_lock@ocean@usml@@UEAAXAEBVwposition@types@3@AEBVseq_vector@53@NV?$matrix@NU?$basic_row_major@_K_J@ublas@numeric@boost@@V?$unbounded_array@NV?$allocator@N@std@@@234@@ublas@numeric@boost@@N2PEAV?$matrix@V?$vector@NV?$unbounded_array@NV?$allocator@N@std@@@ublas@numeric@boost@@@ublas@numeric@boost@@U?$basic_row_major@_K_J@234@V?$unbounded_array@V?$vector@NV?$unbounded_array@NV?$allocator@N@std@@@ublas@numeric@boost@@@ublas@numeric@boost@@V?$allocator@V?$vector@NV?$unbounded_array@NV?$allocator@N@std@@@ublas@numeric@boost@@@ublas@numeric@boost@@@std@@@234@@89boost@@@Z)
2>ocean_shared_test.obj : error LNK2001: unresolved external symbol "public: virtual void __cdecl usml::ocean::boundary_lock::scattering(class usml::types::wposition const &,class usml::types::seq_vector const &,double,class boost::numeric::ublas::matrix<double,struct boost::numeric::ublas::basic_row_major<unsigned __int64,__int64>,class boost::numeric::ublas::unbounded_array<double,class std::allocator<double> > >,double,class boost::numeric::ublas::matrix<double,struct boost::numeric::ublas::basic_row_major<unsigned __int64,__int64>,class boost::numeric::ublas::unbounded_array<double,class std::allocator<double> > >,class boost::numeric::ublas::matrix<class boost::numeric::ublas::vector<double,class boost::numeric::ublas::unbounded_array<double,class std::allocator<double> > >,struct boost::numeric::ublas::basic_row_major<unsigned __int64,__int64>,class boost::numeric::ublas::unbounded_array<class boost::numeric::ublas::vector<double,class boost::numeric::ublas::unbounded_array<double,class std::allocator<double> > >,class std::allocator<class boost::numeric::ublas::vector<double,class boost::numeric::ublas::unbounded_array<double,class std::allocator<double> > > > > > *)" (?scattering@boundary_lock@ocean@usml@@UEAAXAEBVwposition@types@3@AEBVseq_vector@53@NV?$matrix@NU?$basic_row_major@_K_J@ublas@numeric@boost@@V?$unbounded_array@NV?$allocator@N@std@@@234@@ublas@numeric@boost@@N2PEAV?$matrix@V?$vector@NV?$unbounded_array@NV?$allocator@N@std@@@ublas@numeric@boost@@@ublas@numeric@boost@@U?$basic_row_major@_K_J@234@V?$unbounded_array@V?$vector@NV?$unbounded_array@NV?$allocator@N@std@@@ublas@numeric@boost@@@ublas@numeric@boost@@V?$allocator@V?$vector@NV?$unbounded_array@NV?$allocator@N@std@@@ublas@numeric@boost@@@ublas@numeric@boost@@@std@@@234@@89boost@@@Z)
2>eigenverb_test.obj : error LNK2019: unresolved external symbol "__declspec(dllimport) public: __cdecl usml::eigenverb::envelope_collection::envelope_collection(class usml::types::seq_vector const *,unsigned __int64,class usml::types::seq_vector const *,double,double,double,unsigned __int64,u [... cut ...] ion "public: void __cdecl eigenverb_test::envelope_basic::test_method(void)" (?test_method@envelope_basic@eigenverb_test@@QEAAXXZ)
2>C:\Users\David Thibaudeau\Documents\usml-build\Release\usml_test.exe : fatal error LNK1120: 3 unresolved externals
3>------ Build started: Project: INSTALL, Configuration: Release x64 ------
3>  -- Install configuration: "Release"
3>  -- Up-to-date: C:/local/usml/include/usml/ublas/math_traits.h
3>  -- Up-to-date: C:/local/usml/include/usml/ublas/matrix_math.h
````

Tibonium, 2015-05-28:
Clean build resolved

commit b77eca08 (2015-05-20): Added needed code to run_envelope_generator.
Many added cout's remain for debugging.
Many hard coded minimizing in envelope_model, and evelope_generator.
Fixed Bug in boundary_lock, adding scattering method's. Resolves #210.
Cleaned up doc in boundary_model.
Cleaned up doc in volume_lock.
closed 2015-05-28 by Tibonium