campreilly/UnderSeaModelingLibrary #218: Over including of headers

Tibonium, 2015-05-28:
I'd like to propose a change in our include schema. I purpose that only header files that are required for a class be included in the class header, rather then including the module header.

Many classes require access to the wposition classes and as such simply include the types module header. However, this inadvertently includes seq_vector classes and data_grid classes. As such when work is being done on data_grid classes, these classes would still be recompiled, but should not have to be as they do not depend on data_grid.

Example: wave_front has no need to know anything about data_grids, only wpositions and seq_vectors, yet it includes the entire types module.

campreilly, 2015-05-28:
I think this idea has merit, but it could be tricky to implement. 

* Eclipse has some existing tools to "Organize Includes" (under the Source menu, But I'm not sure that all of the choices that it makes are good ones.When one class includes another, and that class includes a third, should the third class be included?  If so, does this choice also apply to items in the standard C++ library?
* We would still have a need to capture the documentation currently provided by the module header.
* It would require kind of a lot of work, Is the benefit adequate to justify that cost?
* What should the impact of this change be on the way that we are currently using namespaces?  If we wanted to get really formal, we'd drop all use of the "using" directive and make everything explicit.  Does that make our code clearer and easier to maintain, or harder?

Tibonium, 2015-05-28:
My proposal is to splinter module headers into micro-headers that are specialized heirachy headers, ie sequences.h would include seq_vector class headers, and then the module header would then include these micro-headers. This preserves the nature and design of the module headers and allows smaller scope including of headers in internal code.

campreilly, 2015-05-28:
If we are going to go through all this effort, I don't think that the micro-module headers have enough added value.  Let's leave it alone for now.

closed 2015-05-28 by campreilly