campreilly/UnderSeaModelingLibrary #219: Modify seq_augment to take in a spacing parameter

islanderjtn, 2015-05-29:
seq_augment at present add num_rays/2 to lower and upper extremes.  The rays are added between the first ray and second ray, and the last ray and second to last ray. The addition rays a equally spaced  between these rays.
After evaluating the fathometer generation it was found that a very small ray spacing at the extremes was needed, such as 0.1 degrees. 
So seq_augment needs to be modified to take in a spacing parameter, and default to 0.1 if not provided. 
So that for a seq_linear of  -90, 90, 181, with 6 addional rays  we would get rays -90.0 -89.9, -89.8, -89.7, -89.0, -88.0, .....  89.0, 89.7, 89.8, 89.9, 90.0.

Tibonium, 2015-05-29:
After discussing with Ted, we have found out that there was a misunderstanding on how and why seq_augment was created. I showed him my regression test that shows the accurate production of all eigenrays utilizing the seq_augment by adding only 4 rays to a seq_rayfan. These results are all produced with the same scenario to where this issue was derived from.

So in conclusion, using a seq_augment with an additional 4+ rays will provide the necessary solution to this issue.

islanderjtn, 2015-05-29:
Not and issue. Not required. seq_augment works as designed.

closed 2015-05-29 by islanderjtn