campreilly/UnderSeaModelingLibrary #226: data_grid_svp bug

JohnAAC, 2015-06-17:
this is from data_grid_svp.h, and I guess you should be testing interp_type(2) in the second condition.

            if ((interp_type(1) != GRID_INTERP_LINEAR)
                    || (interp_type(1) != GRID_INTERP_LINEAR)) {
                interp_type(1, GRID_INTERP_LINEAR);
                interp_type(2, GRID_INTERP_LINEAR);
            }

JohnAAC, 2015-06-17:
Also, in the detailed comments there is a reference to "both axes" should have the same interp type.  But there are three axes.  The comment should be clarified.

campreilly, 2015-06-30:
Do we really need the if blocks on these statements in the first place?
```
            if (interp_type(0) != GRID_INTERP_PCHIP) {
                interp_type(0, GRID_INTERP_PCHIP);
            }
            if ((interp_type(1) != GRID_INTERP_LINEAR)
                    || (interp_type(1) != GRID_INTERP_LINEAR)) {
                interp_type(1, GRID_INTERP_LINEAR);
                interp_type(2, GRID_INTERP_LINEAR);
            }
```

JohnAAC, 2015-06-30:
Designation: Non-Finmeccanica
Seems kind of redundent.

3.3.042312

campreilly, 2015-06-30:
We discussed it here an agreed that they can come out.  I'll start working on this now in the release 1.0 branch.

commit fe1c611a (2015-06-30): Resolve #226 - data_grid_svp fails to set interp_type
- Removed redundant "if" blocks from around the setting of interp_type().
closed 2015-06-30 by campreilly