campreilly/UnderSeaModelingLibrary #231: envelope_generator::beam_gain() maytonly work for single beams

campreilly, 2015-07-01:
In the std::copy() below, the result always seems to be put into beam_matrix.begin1().  How do the second and subsequent indices get filled in?

```
matrix<double> envelope_generator::beam_gain(
    sensor_params::beam_pattern_list beam_list,
    const seq_vector* freq, double de_rad, double az_rad, orientation orient)
{
    // beam_level requires ublas vector
    const vector<double> frequencies = *freq;

    matrix<double> beam_matrix( freq->size(), beam_list.size() ) ;

    vector<double> level( freq->size(), 0.0 ) ;

    BOOST_FOREACH( beam_pattern_model::id_type id, beam_list) {

        beam_pattern_model::reference bp = beam_pattern_map::instance()->find(id);
        bp->beam_level(de_rad, az_rad, orient, frequencies, &level ) ;

        std::copy(level.begin(), level.end(), beam_matrix.begin1());
    }

    return beam_matrix;
}
```

Tibonium, 2015-07-01:
std::copy(source.begin(), source.end(), dest.begin())<http://en.cppreference.com/w/cpp/algorithm/copy>

The information contained in this message, and any attachments, may contain privileged and/or proprietary information that is intended solely for the person or entity to which it is addressed. Moreover, it may contain export restricted technical data controlled by Export Administration Regulations (EAR) or the International Traffic in Arms Regulations (ITAR). Any review, retransmission, dissemination, or re-export to foreign or domestic entities by anyone other than the intended recipient in accordance with EAR and/or ITAR regulations is prohibited.

campreilly, 2015-07-01:
Sorry - I wasn't clear.  I think the point is that the destination always seems to point to beam_matrix.begin1().  Shouldn't the destination be different for each value of ```id```?  To me, it looks like we need to add an iterator for the the destination, but I could be wrong.

campreilly, 2015-07-06:
Fixed a typo in title.

closed 2024-01-03 by campreilly