campreilly/UnderSeaModelingLibrary #252: Are there any thorough examples or documentation of how to use this library, particularly WaveQ3D

factoidforrest, 2023-06-23:
I'm just not sure from the very limited html documentation how one uses this library. How would one go about loading in a 3d world and running a simulation to produce sound?

I see a lot of MatLab, would that be a typical way? Is there an example laying around here somewhere of an end to end workflow?

campreilly, 2023-06-24:
The USML library is primarily designed to be used as a shared library in a larger underwater acoustic simulation that needs rapid calculations of multi-static, broadband transmission loss and reverberation for a collection of discreet sensors and targets (ex: sonar training systems). There is no “user interface” or “run streams”. The Matlab routines are only there to provide visualization of unit test results, usually saved to netCDF files. But you can also create a standalone C++ application like we do in studies/ray_speed.Note that the ocean package provides C++ programmers with abstract interfaces that allow them to wrap your own data sources in a form that the rest of USML can understand. Note that part of making WaveQ3D fast for discreet sensors and targets makes it reallly slow for full field computations with thousand and thousands or range/depth combinations. Very focused on trying to solve its core problem than trying to be all things to all people. Good luck,Dr. Sean M. Reilly

campreilly, 2023-06-24:
FYI - The build process has an option to generate application programmer’s interface (API) documentation in HTML from the source code using doxygen.

factoidforrest, 2023-06-26:
Thank you for the reply, that's helpful.

 I assumed that any documention that could be generated with doxygen would be committed already. 

Reading the /studies directory has been very helpful and I think it gives me the start I need to produce my data. 

I am attempting to get the build system working and if I do, I'll publish a containerized build environment so that others can also easily build it. Building old C projects on various distros is such a tough one.

factoidforrest, 2023-06-28:
Okay here we go, here is a container that has everything to build USML. 

https://hub.docker.com/repository/docker/light24bulbs/usml-build/general
That should make building FAR easier for anyone else. Getting boost going was especially tricky, but it was possible by building boost with `./b2 --layout=tagged` which produces the file names this project expects. 

I'd also like to host a statically linked copy of USML so that nobody has to go through that again when they just want to use usml. I'm not sure of the licenses but I'll look into that and give it a shot.

campreilly, 2023-06-29:
The reason that you are struggling to create a sonogram is because we never got to the digital signal processing (DSP) part in the USML library itself. All the applications that we were working with already had a fully suite of DSP stuff and just needed USML to provide better environmental effects. I tell you how pound the noise is, where it is in the beam pattern, and travel time, but I have no ability to make time series acoustic data.  If someone wants to take on DSP signal generators, I’d be happy for the help.

factoidforrest, 2023-07-02:
Got it, that's really helpful! I was hunting around for it and thinking perhaps I just didn't know the terminology. Makes sense.

Are those softwares you're referring to all proprietary/military? Would any of them happen to be affordable or available for scientists?

campreilly, 2023-07-14:
I'm going to close this issue, just because there don't seem to be any more actions required on my part.

closed 2023-07-14 by campreilly