campreilly/UnderSeaModelingLibrary #263: "git-lfs" undocumented requirement

cconvey, 2024-06-15:
The `usml_install.html` file should mention that the `git-lfs` command must be installed before cloning the repository from Github.

It might make sense to also document this in `README.md`, since would-be users are more likely to read that page on the Github website prior to cloning the repository.

**Background**

Some of the repository's NetCDF files ("*.nc") are stored via Git LFS.

``` bash
$ git lfs ls-files -s
6b4777f5b9 * data/bottom_province/sediment_test.nc (1.6 KB)
9b9552790c * data/woa09/salinity_annual_1deg.nc (60 MB)
a265fc2bd9 * data/woa09/salinity_monthly_1deg.nc (597 MB)
9c5b0c0e71 * data/woa09/salinity_seasonal_1deg.nc (274 MB)
34430d65ac * data/woa09/temperature_annual_1deg.nc (60 MB)
8c1f8fd7ef * data/woa09/temperature_monthly_1deg.nc (597 MB)
1b96060773 * data/woa09/temperature_seasonal_1deg.nc (274 MB)
71f0c82502 * netcdf/test/flstrts_bathymetry.nc (816 KB)
279f6a7a41 * netcdf/test/flstrts_temperature.nc (14 KB)
fed91cb20e * studies/cmp_speed/std14bathy.nc (15 KB)
8f6af3d4bc * studies/cmp_speed/std14profile.nc (112 KB)
```

Because of how Git LFS works, the "git lfs" command must be already installed on a system when cloning the repository from Github.  Otherwise those files won't be in their intended state, causing programs like `usml_test` to fail with misleading / confusing error messages.

campreilly, 2024-06-15:
Note that several big changes occurred in the 3.0 release candidate 1 (rc1) and I haven’t had a lot of collaborators to help me find them. 

I absolutely agreed that the documentation should be fixed in the way that you suggest. Are you planning to start a fork and fix some of these issues yourself? Do the files get checked out correctly if you setup lfs correctly?

cconvey, 2024-06-16:
> I absolutely agreed that the documentation should be fixed in the way that you suggest. Are you planning to start a fork and fix some of these issues yourself?

Here's the approach I was thinking to use, unless you'd prefer something different:

I'll create Issues in this repository (i.e., https://github.com/campreilly/UnderSeaModelingLibrary) for any bug reports,  feature requests, etc. that I think you'd want to consider.

For a given Issue, [my fork of the repo](https://github.com/cconvey/UnderSeaModelingLibrary) my or may get a commit that addresses the Issue.  This will mostly hinge on how much my work needs a solution to that Issue.

Then, on a per-Issue basis, we can decide about creating a PR to upstream my solution into the this repo.  (Note that creating a PR doesn't mean we're 100% on the same page regarding the form of the solution.  The PR just gives us a place to work out the kinks.)

> Do the files get checked out correctly if you setup lfs correctly?

I think so. 

I _think_ that having "git-lfs" installed prior to cloning the repo is sufficient, but I'm not sure it's the only valid approach.

E.g., I wonder if this sequence of steps also works:
1. On a system that doesn't have git-lfs installed, clone the repo from Github.
2. Install git-lfs.
3. cd into the checked-out repo and then run `git lfs fetch` or `git lfs pull`.

I haven't tested this because I didn't want to un-install "git-lfs" to try it out, but it should be easy to check.

campreilly, 2024-06-16:
Scheme seems good.  I’ve been worried that even with git-lfs, that you still didn’t have good data files to work with. And it sounds like you are past that point. 

Feel free to suggest new ways of doing business (e.g., clang compiling). Bon chance.

campreilly, 2024-06-20:
Added these warning to README.txt

Warnings:
- The big data files distributed with USML require you to install
  the git large file system (lfs) befor they can be correctly
  downloaded form GitHub.  This is a GitHub requirement.
- The code must be checked out into a directory called "usml"
  in order to build properly.  The build process treate the parent
  of the "usml" as the root of the project.

Fixed in pull request https://github.com/campreilly/UnderSeaModelingLibrary/pull/266 Import changes from GUWA project.

commit 1c47ef2e (2024-06-20): Resolves #263 "git-lfs" undocumented requirement.
Resolves #269 reverb_analytic_test fails to compile
Resolves #265 Invalid array index in data_grid_svp.h
commit d4e14092 (2024-06-21): Import changes from GUWA project. (#266)

- Import changes from GUWA project.
- Derive bp_gaussian beam pattern from bp_solid.
- Add -fPIC to USMLUse.cmake file
- Change platform IDs from int to uint64_t to better support JSE/GRID ids.
- Ignore -Wmaybe-uninitialized from boost::geomentry in eigenverb_collection.h
- Add ability to read USML_DATA_DIR from environment variables in ocean_utils::make_basic() function.
- Add is_acoustic_target flag to platform_model so that it can be skipped in wavefront generation.
- Fix index calculation in rvbts_collection.cc file.
- Add ability to read time_seri [... cut]
closed 2024-06-20 by campreilly