campreilly/UnderSeaModelingLibrary #264: Bad-index error in `rvbts_test`

cconvey, 2024-06-15:
Using a Debug build of USML, the `rvbts_test` produced the following error output:
``` 
$ ./usml_test --run_test=rvbts_test
Running 1 test case...
=== rvbts_test: update_envelope ===
add sensor site1 (36,16,-100)
add sensor site2 (36,16,-500)
task #0 wavefront_generator: site1 for 8 secs
task #1 wavefront_generator: site2 for 8 secs
task #1 wavefront_generator: done
task #0 wavefront_generator: done
task #2 biverb_generator: site1 -> site2
task #2 biverb_generator: done
task #3 rvbts_generator: site1 -> site2
Check failed in file /usr/include/boost/numeric/ublas/functional.hpp at line 1419:
j < size_j
Uncaught exception in thread_task: bad index
...
```

One aspect of the problem is found on this line:

https://github.com/campreilly/UnderSeaModelingLibrary/blob/226e335b1c17df66af9122b0c9111edc2065a0e5/rvbts/rvbts_collection.cc#L122

When the problem occurs, `n` and `first` both have the value `0`.  Because the expression has type _**unsigned** long_,  `t`'s value becomes `18446744073709551615` (i.e., the value produced by `0ul - 1`).

cconvey, 2024-06-16:
I think you need to build USML in Debug mode to see this runtime error.

Boost's checks the runtime value of these indices via its `BOOST_UBLAS_CHECK` macro, which is enabled in pretty much the same circumstances as C++'s `assert` macro.

campreilly, 2024-06-20:
Fixed in pull request #226 Import changes from GUWA project.

closed 2024-06-20 by campreilly