campreilly/UnderSeaModelingLibrary #265: Invalid array index in `data_grid_svp.h`

cconvey, 2024-06-15:
In the `data_grid_svp` class, a local variable is declared as an array of size 2:

https://github.com/campreilly/UnderSeaModelingLibrary/blob/226e335b1c17df66af9122b0c9111edc2065a0e5/types/data_grid_svp.h#L157

Later in that same method body, an invalid index into that array is used:

https://github.com/campreilly/UnderSeaModelingLibrary/blob/226e335b1c17df66af9122b0c9111edc2065a0e5/types/data_grid_svp.h#L217

That function also contains a loop which (based on my reading of the code) illegally *writes* to `offset[2]` during the loop's final iteration:

https://github.com/campreilly/UnderSeaModelingLibrary/blob/226e335b1c17df66af9122b0c9111edc2065a0e5/types/data_grid_svp.h#L169-L208

campreilly, 2024-06-15:
Take a look at issue #257. I’m having some trouble with this class as well. The original purpose of this class was to speed up the calculation by avoiding the recursion implemented in gen_grid class. Basically hardcode one of our more commonly used interpolations instead of using the generic, all purpose algorithm. 

Note that I just recently made a change to use only “double” values in my grids instead of setting the payload type in using a template parameter. I’d be open to revisiting that decision if it made interpretation run faster.

campreilly, 2024-06-20:
Fixed in pull request #226 Import changes from GUWA project.

commit 1c47ef2e (2024-06-20): Resolves #263 "git-lfs" undocumented requirement.
Resolves #269 reverb_analytic_test fails to compile
Resolves #265 Invalid array index in data_grid_svp.h
commit d4e14092 (2024-06-21): Import changes from GUWA project. (#266)

- Import changes from GUWA project.
- Derive bp_gaussian beam pattern from bp_solid.
- Add -fPIC to USMLUse.cmake file
- Change platform IDs from int to uint64_t to better support JSE/GRID ids.
- Ignore -Wmaybe-uninitialized from boost::geomentry in eigenverb_collection.h
- Add ability to read USML_DATA_DIR from environment variables in ocean_utils::make_basic() function.
- Add is_acoustic_target flag to platform_model so that it can be skipped in wavefront generation.
- Fix index calculation in rvbts_collection.cc file.
- Add ability to read time_seri [... cut]
closed 2024-06-20 by campreilly