campreilly/UnderSeaModelingLibrary #284: Bathymetry normals become NaN for steep slopes, resulting in segfault

fullmetalcoding, 2024-08-22:
https://github.com/campreilly/UnderSeaModelingLibrary/blob/6c7f557741daa2f00dc5d9d84dd060437e53f442/ocean/boundary_grid.h#L112

Line 112 in boundary_grid.h seems to generate NaNs for surface normals when the bottom slope shows a very small horizontal component.  The bottom normal is supposed to be a normalized vector with magnitude 1, so it calculates the final component as the square root of one minus the squares of the other two components. Except in the problematic steep locations, the other two components both have values like 0.8, so their squares sum to more than 1 and we try to take the square root of a negative number, yielding NaN. 

These NaNs further go on to cause a segfault in https://github.com/campreilly/UnderSeaModelingLibrary/blob/6c7f557741daa2f00dc5d9d84dd060437e53f442/ocean/profile_grid.h#L109 when it attempts to interpolate the NaNs. 

This behavior was found with WOA23 bathymetry loaded as netcdf. 

Attached is an image of the location of the run (Turks and Caicos, however it has also been replicated off Florida, Vietnam, and the Persian Gulf). Red dots show areas where the bathymetry normals are NaN. The image was created by reading back bathymetry from USML and overlaying where normals are found to be NaNs. 

[image]

campreilly, 2024-08-22:
I appreciate the work you are doing finding errors.  If you'd like to contribute potential solution for these issues, I'd love to have you as a collaborator.  Just send a pull request from your fork.

fullmetalcoding, 2024-08-26:
> I appreciate the work you are doing finding errors. If you'd like to contribute potential solutions for these issues, I'd love to have you as a collaborator. Just send a pull request from your fork.

I'm not quite sure what the mathematical intent is, so unfortunately I can't recommend a fix. I know you're calculating a surface normal, but I'm unfamiliar with doing so in polar coordinates with angles (I generally would just take the cross product between two vectors in the plane).

campreilly, 2024-08-27:
I just wanted you to feel at home in this repo. Please keep the comments coming. Sean Reilly

fullmetalcoding, 2024-08-29:
> I appreciate the work you are doing finding errors. If you'd like to contribute potential solutions for these issues, I'd love to have you as a collaborator. Just send a pull request from your fork.

We came up with the following candidate fixes to the code: 
[image]

On line 109 of boundary_grid.h.

This is essentially an expansion of the cross product to spherical coordinates. This got the code further along, but still resulted in a crash due to a divide by zero in the reflection model code. 

That was fixed with:
[image]

On line 63 of reflection_model.cc 

(We did both vectorized and non vectorized versions of the functions). 

Essentially it checks if N is equal to zero and sets the bottom normal to {1,0,0} if so to prevent the divide by zero. 

However, although this resolves the crashes, we're getting negative proploss for all the eigenrays, so I assume our math is incorrect.

campreilly, 2024-08-31:
Note that the original mathematical derivation can be found at https://digitalcommons.uri.edu/oa_diss/441/.

campreilly, 2024-09-03:
A couple of quick questions to help me get to the bottom of this quickly:

- Can you provide Git SHA tag for the version you are using.  Just to make sure that we are all on the same page.
- When you say that you are using WOA23 bathymetry, what do you mean?  Which files and which fields?
- Lat/long box for your Turks and Caicos example above?

campreilly, 2024-09-03:
> Note that the original mathematical derivation can be found at https://digitalcommons.uri.edu/oa_diss/441/.

On further examination, that derivation doesn't cover this particular part. Sorry.

fullmetalcoding, 2024-09-03:
> A couple of quick questions to help me get to the bottom of this quickly:
>
> * Can you provide Git SHA tag for the version you are using.  Just to make sure that we are all on the same page.
> * When [...]

Apologies. I misspoke in the issue. 

We're using NOAA WOA23 for salinity and temperature profiles with March as the month (and spring as the seasonal). We're using GEBCO for the bathymetry. Both are in NetCDF format and use the netCDF load functions. 

The GEBCO tile for the Turks run was centered on 18.21 N, 70.852 W with 5 degrees in either direction (so a 10 degree x 10 degree square tile). It was an extraction from the GEBCO website. 

We've confirmed that the 2.0.0dr1 tag produces valid (i.e. non crashing, non NaN) results. The problems we're seeing are with the SHA e53f442 tip of master with a most recent commit of Aug 6.

campreilly, 2024-09-03:
The confirmation that it used to work correctly is very good news. I'll continue to investigate.

campreilly, 2024-09-04:
I'm having some trouble replicating this problem on my system.  Please checkout the branch called **issue284-bathy-normals** and re-try your tests. I added a test data file for your Turks and Caicos area.  Then, I added a test called **gebco_slope_test** to the **boundary_test** of the **usml_test** executable. That test loads depth and surface normal values from the boundary_grid that holds the GEBCO data, and tests to see if any are null.  It fails to fail.

I'm still working on plotting routines to replicate your drawing above.

Interestingly enough, I did find a problem on line #138 where is says **switch (2)** instead of **switch (NUM_DIMS)**. But that doesn't seem to effect this test, because I only test the 2D case.

fullmetalcoding, 2024-09-05:
> I'm having some trouble replicating this problem on my system. Please checkout the branch called **issue284-bathy-normals** and re-try your tests. I added a test data file for your Turks and Caicos [...]

We'll look into it. Are there any API changes between 2.0.0 and 3.0.0 that we might need to be aware of that could cause our issues?

campreilly, 2024-09-05:
There are lots of API changes between 2.x and 3.x to support the goal of making shared oceans immutable. Lots of things that used to be pointers are now shared pointers (see release notes). Hence the major version number change to indicate incompatibility with prior versions. And 3.x is still not yet formally unreleased. What we have so far is a development release so that smart people like you can help me fix it. I was unaware that people were active in this version of repo.

campreilly, 2024-09-05:
I meant not yet formally released.

fullmetalcoding, 2024-09-06:
> I'm having some trouble replicating this problem on my system. Please checkout the branch called **issue284-bathy-normals** and re-try your tests. I added a test data file for your Turks and Caicos [...]

We were able to get your unit test to fail by modifying this part https://github.com/campreilly/UnderSeaModelingLibrary/blob/issue284-bathy-normals/ocean/test/boundary_test.cc#L346
 

To instead be
 

for(double latitude = 13.22; latitude < 23.20; latitude += 0.001) {
        for(double longitude = -75.85; longitude < -65.86; longitude += 0.001) {
 

The dev working on it thinks the original loop wasn't checking enough interpolated points
 

Though with a step of 0.001 it also takes forever to run...

campreilly, 2024-09-07:
Can you put a break on the first time that it fails and see if it consistently fails at that lat/long if we just skip the loops?

fullmetalcoding, 2024-09-09:
> Can you put a break on the first time that it fails and see if it consistently fails at that lat/long if we just skip the loops?

13.632, -69.349 is the initial lat long, and disabling the loop does indeed produce a NaN.

campreilly, 2024-09-10:
Have you had aa chance to use the debugger to walk the code and figure out where things are going wrong at that lat/long?

campreilly, 2024-09-10:
Yay!  Breaks on my machine as well!

=== boundary_test: gebco_slope_test ===
reading /home/sreilly/Projects/usml/data/bathymetry/gebco_2024_n23.21_s13.21_w-75.852_e-65.852.nc
writing tables to /home/sreilly/Projects/usml/ocean/test/gebco_slope_test.csv
/home/sreilly/Projects/usml/ocean/test/boundary_test.cc(367): [1;31;49merror: in "boundary_test/gebco_slope_test": check !std::isnan(normal1.rho()) has failed[0;39;49m
/home/sreilly/Projects/usml/ocean/test/boundary_test.cc(371): [1;31;49merror: in "boundary_test/gebco_slope_test": check !std::isnan(normal.rho(0, 0)) has failed[0;39;49m
/home/sreilly/Projects/usml/ocean/test/boundary_test.cc(378): [1;31;49merror: in "boundary_test/gebco_slope_test": difference{nan} between normal1.rho(){-nan} and normal.rho(0, 0){-nan} exceeds 1e-06%[0;39;49m

campreilly, 2024-09-10:
Just some useful intel in debugging this program.
```
ncks -O --no_abc -d lat,13.6145833333333,13.63125 -d lon,-69.36874,-69.34375 gebco_2024_n23.21_s13.21_w-75.852_e-65.852.nc test.nc
ncdump test.nc >test.txt

 lat = 13.6145833333333, 13.61875, 13.6229166666667, 13.6270833333333, 
    13.63125 ;

 lon = -69.3645833333333, -69.3604166666667, -69.35625, -69.3520833333333, 
    -69.3479166666667, -69.34375 ;

 elevation =
  -5018, -5015, -4978, -4895, -4804, -4743,
  -5011, -5007, -5004, -4857, -4691, -4598,
  -5009, -5005, -5007, -4805, -4528, -4447,
  -5007, -5005, -5010, -4793, -4304, -4304,
  -5006, -5006, -5013, -5011, -4319, -4304 ;
}
```

campreilly, 2024-09-10:
We are sort of back to where we started. The data at 13.632, -69.349 beautifully illustrates the problem in your original write-up.  But now I'm in a position to actually see it break and fix it on my computer.  Just as soon as I figure out how to fix the theory.  Like you originally said, I think it is all in the conversion from the bottom slope in the latitude and longitude directions into the surface normal relative to at the specified location on the interface.

campreilly, 2024-09-11:
I think we are using the wrong formula to compute the surface normal.  Look at better derivation in https://mathworld.wolfram.com/NormalVector.html

fullmetalcoding, 2024-09-11:
> BTW - Who is `fullmetalcoding` and why are they interested?
> 
> I am Dr. Sean M. Reilly, formerly of University of Rhode Island ([campreilly@gmail.com](mailto:campreilly@gmail.com)) and now acting as Technical Director - Live, Virtual, and Constructive Solutions Group, Mission Technologies, Huntington Ingalls Industries (HII) (sean.reilly@hii-tsd,com). We build sensor models for real-time distributed simulation systems.

I'm the acoustic simulations lead at Naval Surface Warfare Center Carderock for SONAR training systems. We were evaluating WaveQ3D for use in some of our simulations. We had been working with 2.0 but wanted to test 3.0 for its 3D reverb capability.

campreilly, 2024-09-11:
The 3D reverb went through a pretty big change and I don't mind saying that I'm a little nervous.  I promise not to formally release 3.0 until you folks are satisfied.

campreilly, 2024-09-11:
Here is a better dataset around 13.632, -69.349.
```
ncks -O --no_abc -d lat,13.6229,13.639589 -d lon,-69.35625,-69.3354 gebco_2024_n23.21_s13.21_w-75.852_e-65.852.nc test.nc
ncdump test.nc >test.txt

 lat = 13.6229166666667, 13.6270833333333, 13.63125, 13.6354166666667, 
    13.6395833333333 ;

 lon = -69.35625, -69.3520833333333, -69.3479166666667, -69.34375, 
    -69.3395833333333, -69.3354166666667 ;

 elevation =
  -5007, -4805, -4528, -4447, -4431, -4400,
  -5010, -4793, -4304, -4304, -4380, -4424,
  -5013, -5011, -4319, -4304, -4435, -4512,
  -5008, -5006, -5008, -4734, -4661, -4661,
  -4998, -5000, -5008, -4902, -4822, -4790 ;
```

campreilly, 2024-09-12:
The interpolation of the height returns the gradients in the theta (co-latitude) and phi (longitude) directions in units of meters per radian. To scale these gradients into unitless values (meters per meter), we must scale them by the meters per radian in the theta and phi directions.

[image]

[image]

The surface normal vector in the $\rho, \theta, \phi$ directions is given by

[image]

normalized to unit length.

[References:](file:///home/sreilly/Projects/usml/html/d5/d55/references.html#_references000016)
    Weisstein, Eric W. "Normal Vector." From MathWorld–A Wolfram Web Resource. https://mathworld.wolfram.com/NormalVector.html

campreilly, 2024-09-12:
@fullmetalcoding I have a prototype change implemented in the issue284-bathy-normals branch.  Not tested yet, but figured that you might want to test in parallel.

campreilly, 2024-09-13:
@fullmetalcoding I could use some help on two topics here:

1. How did that ever work? - You say that this works fine in the 2.0 version, but I can't see how that's possible give the difference in the math used (include +1 in N and divide all terms by N not just rho).  Do you mean that it never crashes or that the calculation of the normal direction was accurate?
2. How do we build a better unit test? - Given the errors you found, how do we create a unit test for the normal calculation that verifies that accuracy of the normal calculation?  Because "by inspection" clearly didn't work as a test methodology last time.  All my ideas so far involve circular logic.

fullmetalcoding, 2024-09-13:
> @fullmetalcoding I could use some help on two topics here:
>
> 1. How did that ever work? - You say that this works fine in the 2.0 version, but I can't see how that's possible give the difference in [...]

It does produce some bad eigenrays, but we filter those out. We had assumed those were due to our targets of interest being on land (we track roughly 1500 targets so we assumed some of them might run aground). 

By "it works in 2.0" I mean it doesn't crash, and it produces some number of non-degenerate eigenrays that give a "physically plausible" spectrum. In 3.0 we got a crash, and then when we "fixed" the crash, all of the eigenrays were degenerate.

campreilly, 2024-09-14:
Any ideas on a great unit test?

fullmetalcoding, 2024-09-16:
> Any ideas on a great unit test?

The only thing I can recommend is try to structure the unit test the way you expect it to be used in the wild. That's how we found the issues we did. 

Our runs basically look like: 
- Set sensor position (lat/lon/depth)
- Set up an SSP (Pick a month, feed WOA23 salinity and temp profiles and seasonal profiles) 
- Set up bathymetry (We usually feed a custom tile pulled from GEBCO) 
- Set the bottom type to basalt (WIP until we incorporate bottom type databases)
- Add up to 1500 target lat longs/depths
- Set up some number of freqs
- Set up 256x128 Azimuths and D/Es (we generate a field that represents power per sterradian at a specific frequency)
- March out to some max time (we generally go to around 100 s)

Other options include things like "data fuzzing" where you try to feed it random data and ensure it works or breaks gracefully. 

I think the reason your existing tests didn't catch what we ran into is just a question of resolution though.

campreilly, 2024-09-16:
Draft Unit Test:
- create a new bathymetry file in the area of Turks and Caicos Islands from the GEBCO website gebco_2024_n20.85_s19.75_w-69.75_e-68.75.nc
- checkout the old version of boundary_grid.h
- run `./usml_test --run_test=boundary_test/gebco_slope_test` from the build directory
   - 129 failures are detected
   - First NaN for the rho component of norm is 19.760417,-69.052083,-2433.000000,-nan,0.770606,-0.799428
- run test_gebco_slope unit test in python/tests/testbathy.py to plot bathymetry depth, pitch, and heading.
   - save "old" copies of the test_gebco_slope_pitch.png and test_gebco_slope_heading.png files
   - check that test_gebco_slope_depth.png is very similar to gebco_2024_n20.85_s19.75_w-69.75_e-68.75_cm.png downloaded from GEBCO
- replace boundary_grid.h with new version and re-run C++ unit tests 
   - No errors detected
-  Re-run test_gebco_slope unit test
   -  save "new" copies of the test_gebco_slope_pitch.png and test_gebco_slope_heading.png files
   - note that the highest "new" pitch levels are slightly less than those of the highest "old" pitch levels 
   - note that there are only slight differences in normal heading between "old" and "new"

gebco_2024_n20 85_s19 75_w-69 75_e-68 75_cm.png
[image]

test_gebco_slope_depth.png
[image]

test_gebco_slope_pitch_old.png
[image]

test_gebco_slope_pitch_new.png
[image]

test_gebco_slope_heading_old.png

[image]

test_gebco_slope_heading_new.png
[image]

campreilly, 2024-09-17:
What kind of speeds are you getting for 1500 targets? That's much more than I typically test with.

PR #285 (2024-09-27): Issue284 Bathymetry normals become NaN for steep slopes, resulting in segfault
Fix the equations for the surface normal, using Weisstein, Eric W. "Normal Vector." From MathWorld–A Wolfram Web Resource. https://mathworld.wolfram.com/NormalVector.html. Built python unit test to illustrate the differences in the old and new algorithms.

The interpolation of the height returns the gradients in the theta (co-latitude) and phi (longitude) directions in units of meters per radian. To scale these gradients into unitless values (meters per meter), we must scale them by the meters per radian in the thet [... cut]
closed 2024-09-27 by campreilly